acconeer/acconeer-python-exploration #1: Measured height is not correct.

zoyaforever, 2019-03-11:
Hi, I run the envelope.py program under the example directory, and change the config to the following:
<pre>
    config = configs.EnvelopeServiceConfig()
    config.sensor = args.sensors
    config.range_interval = [0.2, 4.0]
    config.sweep_rate = 60
    config.gain = 0.95
    config.session_profile = configs.EnvelopeServiceConfig.MAX_SNR
    config.running_average_factor = 0.95
</pre>

[image]

I put the sensor board(XM112 & XB112) above the ground about 750mm, but the result from envelope  is about 2.6 m, so did I do the wrong configs?

erikmansson, 2019-03-11:
Hi @zoyaforever! Thank you for getting in touch with us.

You aren't doing anything wrong - it's our fault. Due to limited buffer sizes, we can only measure a total range of about 0.7-0.9m in one go. If a longer range is set, it will be shortened to the maximal length. Currently, this is *not* reported to the user. I will commit an update later today that allows you to see what range was actually set. It will be seen in the info dictionary which is returned from setup_session(...).

We are working on supporting longer sweeps by "stitching" several subsweeps together. This is already supported in the main software, but needs some work before we can use it in the Exploration tool. We are releasing experimental support for the Raspberry Pi EVK:s soon.

erikmansson, 2019-03-11:
Now I've updated the client so that it reports if the measured range differs from the one you requested. See commit 1e258ca.

zoyaforever, 2019-03-12:
@erikmansson  Thanks for you help.

I have another question:
Is the buffer limit caused by A111 sensor or the ATSAME70Q21A mcu?

erikmansson, 2019-03-12:
No problem!

The limiting buffer is in the A111 sensor itself.

erikmansson, 2019-03-18:
I'll close this issue now. Please feel free to reopen it or create a new issue if you have further questions.

closed 2019-03-18 by erikmansson

misisnik, 2019-03-18:
:(

erikmansson, 2019-03-20:
As of commit 546de1c, there is now support for experimental stitching in the GUI when using Raspberry Pi based EVK:s. This is only supported for SW 1.7.1 (or newer).

zoyaforever, 2019-03-21:
@erikmansson   Thanks

zoyaforever, 2019-05-12:
hi @erikmansson 
I have a question about integrate A111 into my own STM32L476 boards, but I find no where to post the question, So I ask you directly here.

In file acc_board_stm32l476.c of SDK(acconeer_stm32l476_freertos_v1_8_1), it has following code:

```c
/**
 * @brief The number of sensors available on the board
 */
#define SENSOR_COUNT 1

/**
 * @brief Host GPIO pin number (BCM)
 *
 * This GPIO should be connected to sensor 1 GPIO 5
 */
#define SENSOR_INTERRUPT_S1_PIN 8 //PA8

/**
 * @brief Host GPIO pin number (BCM)
 */
/**@{*/
#define RSTn_PIN   9  //PA9
#define ENABLE_PIN 21 //PB5

#define PMU_ENABLE_PIN 10 //PA10
#define SPI_CS_PIN     22 //PB6

#define CE_A_PIN 20 // PB4
#define CE_B_PIN 26 // PB10
```

According the HAL - User Guide, I know what SENSOR_INTERRUPT_S1_PIN, ENABLE_PIN and SPI_CS_PIN    means, but what is the meaning of following?
- RSTn_PIN   
- PMU_ENABLE_PIN 
- CE_A_PIN 
- CE_B_PIN 

And is there any place where I can ask questions about SDK?

Thanks.

erikmansson, 2019-05-12:
Sorry, this is not my area of expertise. If you send your question to info@acconeer.com our customer support team will help you. I'll make sure it reaches them tomorrow :)

zoyaforever, 2019-05-12:
Ok, thanks you very much.

erikmansson, 2019-05-27:
@zoyaforever as of 927c3fb, there is now limited experimental stitching for the XM112 :smile:

zoyaforever, 2019-05-27:
@erikmansson ok, thanks. I will try it.