acconeer/acconeer-python-exploration #2: Getting this working on Pi 3 B+

nerble, 2019-04-21:
Get this issue while running under a virtualenv (or not under one, for that matter):
`python -m pip install --user -r requirements.txt`

```
Collecting PyQt5 (from -r requirements.txt (line 8))
  Could not find a version that satisfies the requirement PyQt5 (from -r requirements.txt (line 8)) (from versions: )
No matching distribution found for PyQt5 (from -r requirements.txt (line 8))
```

Using python 3.4.3 which seems to be the highest version I can get operating on this Pi under conda / stretch.  Any advice?  Anyone have a guide to getting this operating on the Pi 3 B+?

erikmansson, 2019-04-21:
We generally do not recommend running the Exploration scripts on a Raspberry Pi. As such, we don't support or test for it. Having that said, you could probably get PyQt5 up and running in some other way. For example, [this](https://raspberrypi.stackexchange.com/questions/62939/pyqt5-on-a-raspberry-pi) post on StackExchange seems to have a promising answer. May I ask why you want to run Exploration on a Pi? Perhaps you don't need PyQt5 at all for your application?

nerble, 2019-04-23:
I'm working on an interface for an embedded system using the X111.  The Exploration scripts looked like a good way to get started.  I've tried multiple versions of the instructions you link to and with and without virtual environments.  It's especially annoying as people claimed they've done it and got it to work and I've only followed their instructions.

I've since installed Ubuntu Mate (18.04) on my Pi but still getting the same error, even with compiling and installing sips and PyQt5 from source.  My Python version is 3.6.7.

erikmansson, 2019-04-23:
I still don't quite understand what you're trying to do. I'm guessing you want to run the GUI and/or standalone plotting examples directly on your Pi, connecting to the streaming server on that same Pi. Am I correct?

Are the other instructions you're following available online?

If you've successfully installed PyQt5 already, then you shouldn't try to install it with pip. Remove PyQt5 from requirements.txt or simply install the remaining dependencies manually.

nerble, 2019-04-25:
Well I got it working, though I don't recommend the experience to anyone else.  And I tried so many things I couldn't write a 'how to,' though building PyQt5 and sip yourself and using Ubuntu Mate is the way to go.  Thanks for your help and suggestions.

closed 2019-04-25 by nerble