acconeer/acconeer-python-exploration #3: Error when using background files

dlabun, 2019-04-22:
After running a background scan and saving the file I receive the following error when a start a normal scan using either IQ or Envelope :

Cannot load clutter (file location). Object arrays cannot be loaded when allow _pickle=False

dj-acconeer, 2019-04-22:
Could you explain the steps you do to reproduce the problem?
Also, what environment are you working in?

dlabun, 2019-04-22:
Windows 10 running Python 3.7.3 and a copy of this project downloaded today. I am using the XM112 module with 1.8.1 loaded on it.

To produce the error just launch the gui/main.py file as per the read me on this profile, you'll immediately see in the console the error message. If you run a background scan with IQ or Envelope selected the background scan will run normally and then prompt you to save the file. Once you to try to start a normal scan there will be a pop up dialog box with the same "Cannot load clutter" error message that's listed in the console.

dj-acconeer, 2019-04-22:
Thanks for pointing out this issue!
It looks like this is related to an update in the numpy module with version 1.16.3, which defaults allow_pickle to False.
I will update the code accordingly.
In the mean time, you can add "allow_pickle=True" to the np.load() call in data_processing.py and main.py to fix the problem (or downgrade the numpy module).

dj-acconeer, 2019-04-23:
Fixed.
See commit 2fc1a37c2abb7971399e8e533aed44a39a295e46

closed 2019-04-23 by dj-acconeer