acconeer/acconeer-python-exploration #4: Slower XM112 UART data rates with 1.8.1 firmware

dlabun, 2019-04-24:
Ever since the 1.8.1 SDK and module firmware were released I've noticed that the python exploration tools are reporting much slower data transfer rates. Prior to updating to 1.8.1 I was normally seeing 3.0Mbit/s being reported. After updating to 1.8.1 it's dropped to 0.18Mbit/s and stays there regardless if I change computers or USB cables.

erikmansson, 2019-04-24:
How are you measuring your data rate? Or is it the reported data rate from the client running with the verbose flag -v or -vv?

dlabun, 2019-04-24:
This is strictly what's being reported in the console window when you launch gui/main.py. On 1.7.1 I always saw a "fast" connection speed with an occasion dropped frame message, maybe once every 15 minutes. With 1.8.1 it looks like the module is automatically downshifting to the slowest speed and never attempt to run any faster.

Now I have not updated the exploration project files on my computer since the allow_pickle fix yesterday... Would that have any bearing on data transfers?

erikmansson, 2019-04-24:
Oh, I see whats happening. You get something like this:
```
16:05:34.555 | INFO    | MainProcess      | acconeer_utils.clients.reg.client    | data rate: 0.18 Mbit/s
```
in the console after hitting connect in the GUI.

What you're seeing is the GUI testing to run a session upon connecting. When a session is started, the client logs what the effective data rate will be for the config used. Since the default config is envelope @ 10 Hz, the data rate will be very low. We have changed the default config, and that's probably why you're seeing this.

If you want to test the actual maximum throughput you can use the script `examples/utils/test_throughput.py`. Try bumping the sweep rate to 100 Hz (line 22) and use the -v or -vv flag. You will see that the config requires 2.65 Mbit/s, but the throughput will be somewhere around 2.0 Mbit/s, which is the maximum. Using the -vv flag, you can also see the baudrate used. If the throughput is not enough, try using SPI.

The allow_pickle fix does not affect data rate.

erikmansson, 2019-05-03:
I'll close this for now. Please feel free to reopen this issue again if you want.

closed 2019-05-03 by erikmansson