acconeer/acconeer-python-exploration #8: ImportError: cannot import name 'QtGui' from 'PyQt4'

chronopoulos, 2019-09-27:
I'm not sure what to do about this: acconeer-python-exploration requires python3.7, and pyqtgraph requires PyQt4, but PyQt4 is not compatible with python3.7; pip3 won't install it, and the version provided by apt, python3-pyqt4, gives the following error:

```
from PyQt4 import QtGui, QtCore, uic
ImportError: cannot import name 'QtGui' from 'PyQt4' (/usr/lib/python3/dist-packages/PyQt4/__init__.py)
```
(Despite the fact that this same line works fine from python3.5)

Any advice, or is this a compatibility bug?

dj-acconeer, 2019-09-27:
Could you explain what exactly you are trying to do?
There are no PyQT4 imports in the code.
In general, PyQtGraph supports PyQt5.

chronopoulos, 2019-09-30:
i see, thanks. to answer your question, i was trying to run:

$ python3.7 examples/services/envelope.py -u /dev/ttyUSB0

as suggested in the [XM112 Quick-Start Video](https://www.youtube.com/watch?v=FjYTIySbLZk). there may not be PyQt4 imports in the acconeer code, but pyqtgraph does import it behind the scenes; and apparently it selects PyQt4 over PyQt5, if available.

i resolved the issue by removing my (global) installation of python3-pyqt4. since the global python3 packages are available to a local python3.7 environment, pyqtgraph was finding PyQt4, importing it, and then suffering some compatibility issue that i still don't understand. that is, why could it import PyQt4, but not the QtGui submodule?

anyway, as my situation is resolved, i will close the issue. thanks for your help!

closed 2019-09-30 by chronopoulos