acconeer/acconeer-python-exploration #9: Wrong ELF class for libft4222 on Raspbian

pferrill, 2019-10-07:
Using a Raspberry Pi 4 I've tried the default Rasbpian install and another 64-bit install and they both cause the gui\main.py app to fail when trying to connect to the SPI with the error libft4222.so.1.4.2.184: wrong ELF class: ELFCLASS64.

Not sure what to try next.

erikmansson, 2019-10-07:
There currently is no support for any ARM architecture. If the [drivers](https://www.ftdichip.com/Products/ICs/FT4222H.html) are compatible, that could probably be fixed.

But let me first ask you what you are trying to achieve. Are you trying to connect to an XM112 module via your Pi?

pferrill, 2019-10-08:
I'm using the SparkFun A111 board and following the instructions provided on their site.

erikmansson, 2019-10-09:
Ok! I can't find exactly what you're referring to, could you give me some pointers?

Normally we run the _streaming server_ on the Pi (located under the utils directory after building), and Exploration on the machine you're working on. As long as the server is running on your Pi, you should be able to connect and stream data from it with Exploration.

pferrill, 2019-10-09:
That works!

I was confused as it mentioned SparkX which I believe is a different product than the A111.

Thanks for your help.

erikmansson, 2019-10-09:
Great! Sorry for the confusion - we should make that more clear.

closed 2019-10-09 by erikmansson