acconeer/acconeer-python-exploration #11: actual measured range differs from the requested

fohnbit, 2019-10-15:
Hi,

when I set the range to **2.00, 2.10** I get the warning and no measuring.
Did I need to change also other settings?

https://github.com/acconeer/acconeer-python-exploration/blob/1f323a2ab0de9ce69a053522c4559e5282cbb6ee/examples/processing/sparse_speed.py#L70

Thank you

erikmansson, 2019-10-15:
I can reproduce the warning (in this case it is expected), but I can't seem to reproduce the error.

Some questions:

- Does it work reliably with the default range?
- What hardware are you using?
- What software version (module software or SDK)?

Also, can you give me the output from the session creation? Add a line with `print(session_info)` after
https://github.com/acconeer/acconeer-python-exploration/blob/1f323a2ab0de9ce69a053522c4559e5282cbb6ee/examples/processing/sparse_speed.py#L39

fohnbit, 2019-10-16:
Hello,

yes, with the default range of 0.30, 0.66 it is working and then I don't get the warning.
I'm using a Raspberry Pi 2
I'm using the module software. I flash with Bossa the software around 3 weeks ago

here ist the complete output:
`Autodetected XB112 on /dev/ttyUSB0

15:47:46.504 | WARNING | MainProcess      | acconeer_utils.clients.base          | old server version - please upgrade server
15:47:46.550 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | experimental stitching on - switching to max freq. mode
15:47:46.871 | WARNING | MainProcess      | acconeer_utils.clients.base          | actual measured range differs from the requested
{'frequency': 200.0, 'actual_range_start': 1.982, 'actual_range_length': 0.18, 'actual_stepsize': 0.180164, 'number_of_subsweeps': 128, 'data_length': 256, 'actual_subsweep_rate': 14939.667}
Press Ctrl-C to end session`

erikmansson, 2019-10-16:
Just to be clear, you're running an XM112 via a Pi? Do you remember which version that you flashed? Probably 1.8.6 or 1.9.1.

The output seems normal, and if it gets to that point correctly you _should_ get data. Maybe if you're running 1.8.6 there could be a compatibility issue.

erikmansson, 2019-10-16:
Could you also send us the traceback from the error?

fohnbit, 2019-10-16:
Hello Erik,

yes, the XM112 is on a PI 2

I`m flashed 1.9.1

I put the XM112 again back to the windows pc and test with the GUI version. Maybe I see there something.

fohnbit, 2019-10-16:
> Could you also send us the traceback from the error?

Which error did you mean? I get no speed result.

fohnbit, 2019-10-16:
I forgot:
Because of the warning "old server version - please upgrade server":

Is there an update?

erikmansson, 2019-10-16:
> Hello Erik,
> 
> yes, the XM112 is on a PI 2
> 
> I`m flashed 1.9.1
> 
> I put the XM112 again back to the windows pc and test with the GUI version. Maybe I see there something.

Ok. Then that should probably not be a problem.

> > Could you also send us the traceback from the error?
> 
> Which error did you mean? I get no speed result.

Oh, maybe I misunderstood. So it's running as it should but you're getting no signal?

> I forgot:
> Because of the warning "old server version - please upgrade server":
> 
> Is there an update?

Yep.

fohnbit, 2019-10-16:
Yes, its running but not get any speed result.

How can I update the server? 
`python3 setup.py install --user`

this of course I run

fohnbit, 2019-10-17:
Good morning,

I update my repositories. I clone yours to my pc and make a sync fork to mine. So I´m able to use always the current branch in my fork.

I connect the XM112 to my windows PC and start "Speed (sparse)" with the default settings:
All fine and working

Then I change:
Start: 1.5m
End: 1.6m
Sweeps per frame: 128

When I start, I get in the commandline the output:
`08:26:06.490 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:26:06.981 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:26:09.142 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.9.1
08:26:09.143 | WARNING | MainProcess      | acconeer_utils.clients.base          | old server version - please upgrade server
08:26:09.590 | INFO    | MainProcess      | acconeer_utils.clients.reg.client    | data rate: 0.60 Mbit/s
08:26:15.837 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:26:15.998 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:26:18.451 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:26:18.936 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | data rate: 3.28 Mbit/s
08:26:18.936 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | data rate might be too high
08:27:45.346 | INFO    | MainProcess      | acconeer_utils.clients.base          | reported version: 1.10.0
08:27:45.347 | WARNING | MainProcess      | acconeer_utils.clients.base          | actual measured range differs from the requested
08:27:45.813 | INFO    | MainProcess      | acconeer_utils.clients.reg.client    | data rate: 0.41 Mbit/s
08:27:45.814 | WARNING | MainProcess      | acconeer_utils.clients.base          | actual measured range differs from the requested
Updated range settings to match session info!
Start 1.500 -> 1.501
End   1.600 -> 1.501
08:29:10.561 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | successfully recovered from corrupt frame
08:29:10.738 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | successfully recovered from corrupt frame
08:29:10.904 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | successfully recovered from corrupt frame
08:29:11.074 | WARNING | MainProcess      | acconeer_utils.clients.reg.client    | successfully recovered from corrupt frame`

It changed the range to "1.5, 1.5"
He show me sometimes a speed, where no movement is in front.

When I change:
Sweep per frame: 512
The gui run stable, but no speed is showing.

Would be very nice, when you can give me the correct settings, to measure speed in a distance of 2.00m - 2.20m (or near this range)

Thank you!

When I ch

fohnbit, 2019-10-17:
update:

I see now, there is v 1.10 available and flashing with bossa.
Seems now the problems are gone :-)

erikmansson, 2019-10-17:
Alright, good! I'll close this then :)

closed 2019-10-17 by erikmansson