acconeer/acconeer-python-exploration #12: Settings for Speed measuring

fohnbit, 2019-10-17:
Hello,

after upgrading to 1.10 the sensor is working fine. I complete almost my project, but I has a problem to set the sensitivity:

**What I need:**
Measuring from 2.00 to 3.00 meter the speed of an object. I want each 10cm the current speed

**What I get current**
I get just 2 speed infos on 2.0m and 2.9m with my settings

**My settings**
```
NUM_FFT_BINS = 128
config.range_interval = [2.00, 3.00]
config.stepsize = 16
config.sampling_mode = configs.SparseServiceConfig.SAMPLING_MODE_A
config.number_of_subsweeps = NUM_FFT_BINS
config.gain = 0.5
config.hw_accelerated_average_samples = 60
# config.subsweep_rate = 6e3

# force max frequency
config.sweep_rate = 500
config.experimental_stitching = True
```

**How I understated the settings**
`NUM_FFT_BINS  * config.stepsize` must not be greater than 2048
`config.sweep_rate` frequency ... a higher value result in a more precise measurement and is better

**My question**
So how I has to set the sensor, to measuring with a distance of 1m (2.0m to 3.0m) in 10cm steps?

https://github.com/acconeer/acconeer-python-exploration/blob/86f16e7e82fd08f4f4525afd0a4d1b1236f2bebb/examples/processing/sparse_speed.py#L70-L80

erikmansson, 2019-10-17:
Hi,

The stepsize does not set the number of points measured. Instead, it sets the distance between the points. A `stepsize` of 1 gives an `actual_stepsize` of 6 cm (roughly). 2 gives 12 cm, 3 gives 18 cm, and so on. Since you want steps of 10 cm, a stepsize of 2 is the closest you will get. This with a range_interval of [2, 3] gives you 9 depths points. The number of depth points times the `number_of_subsweeps` gives you the buffer usage (`data_length`), in this case 9 * 512 = 4608.

To solve this you could use
```
range_interval = [2.04, 2.90]
stepsize = 2
NUM_FFT_BINS = 256  # = config.number_of_subsweeps
```

`config.sweep_rate` is set to 500 in conjunction with `config.experimental_stitching = True` to achieve the highest possible sweep rate. It will be much lower than 500, but still the maximum possible. This will be easier to set in the future. Your actual sweep rate will be roughly `actual_subsweep_rate / number_of_subsweeps`.

It can be difficult to see your cat running at these distances. Unfortunately their stealthiness also applies to their radar cross section, which means that the reflected signal will be much weaker than from a human. Try adjusting the gain setting, and/or decreasing the detection threshold:

https://github.com/acconeer/acconeer-python-exploration/blob/86f16e7e82fd08f4f4525afd0a4d1b1236f2bebb/examples/processing/sparse_speed.py#L152

fohnbit, 2019-10-17:
Hello Erik,

ok will try different values at gain and treshold.

I guess I has to increase the gain, to have better chances to see the cats?

erikmansson, 2019-10-17:
Probably yes. But increase it in small steps as higher gain worsens the signal quality.

erikmansson, 2019-10-17:
And oh, of you aren't already, using a lens would probably help a lot too.

You can get a kit [here](https://www.digikey.se/products/en/development-boards-kits-programmers/accessories/783?k=acconeer) or 3D print one yourself. We can supply you with the drawings.

fohnbit, 2019-10-18:
thank you .. I ordered the lens kit and hope for a better result in the detection.

fohnbit, 2019-10-18:
Hello,

I optimized my tests now ... the cat´s are detected in a distance of 2.0m - 2.8m ... but:
When the cat is lying in the range and the sensor detect her: the cat just need to move the head, and I have a high PSD (dB) .. around 12-14

But when they running (or moving faster) through the range, the sensor detect them not.

For me it seems, the sensor need a **minimum** time to detect an object. If he detect the new object, it is **much more sensible** as when a new object come new in the range.
(hope you understand what I try to say)

Because just moving the head of a cat is much less object size to detect, as when the whole cat running through the range.

So is it possible, to get a faster recognizing of new objects?

My settings:
```
config.range_interval = [2.04, 2.76]
config.stepsize = 3
config.sampling_mode = configs.SparseServiceConfig.SAMPLING_MODE_A
config.number_of_subsweeps = 256
config.gain = 0.6
config.hw_accelerated_average_samples = 60
config.sweep_rate = 200
config.experimental_stitching = True
```

and
```
self.fft_length = (self.num_subsweeps // 2) * FFT_OVERSAMPLING_FACTOR
self.num_noise_est_bins = 3
noise_est_tc = 1.0
self.min_threshold = 2.5
self.dynamic_threshold = 0.1
```

erikmansson, 2019-10-21:
There _shouldn't_ be a big difference between fast and slow moving objects.

Do you see anything at all in the PSD when the cat is running past the detection range?

fohnbit, 2019-10-22:
Good morning Erik,

well, I received the lenses now and use it in my temporary box ... the result are much much better .. I use for the first test the HBL Lens in D1 (30°)

I could now made test with a faster speed of the cats. Maybe I use later in D2 (15°). But look now very good. Thank you for your support!

erikmansson, 2019-10-22:
Great!

We haven't looked that much into this problem. There is a very simple idea on how to handle this - the `dynamic_threshold`. It sets the threshold based on the highest detected peak in the PSD. Try raising it and see if that works.

https://github.com/acconeer/acconeer-python-exploration/blob/d8291935f465ffb54ae9a3e3baf7978ab7cd8b52/examples/processing/sparse_speed.py#L154

fohnbit, 2019-10-26:
Thank you @erikmansson 

Just another question to get a faster result:
to check the speed I thought to observe a range of 1m. But this is maybe not optimal.
Maybe a range of 20cm is more efficient (Performance) as 100cm?

I get 2 successful snapshots, but as the cat running 20,4km/h, he passed already the camera. So a faster speed detection would be good.

Attached the capture with 16,8km/h
[image]

erikmansson, 2019-10-26:
That's awesome! Hope you don't mind me sharing with the company :)

Yes, if you want more a precise trigger, shorten the range length. By doing so, you get a higher update rate and make sure that the trigger happens at a more precise depth.

Note that it's only the range length that matters. If it starts at 20cm or 100cm makes no difference. I suspect that it might be best to only use one depth point, i.e., range_start = range_end.

fohnbit, 2019-10-27:
Hi Erik,

thank you and no problem. You can share the pic :-)
I will test next week the new settings.

closed 2019-11-05 by erikmansson