acconeer/acconeer-python-exploration #13: gui and history less sparse_speed.py version

fohnbit, 2019-10-18:
Hello,

I used a GUI and history less version of the sparse_speed.py file. Because running just on a PI and optimize performance.

Now the problem:
With my version I have bad results, with the original GUI version I have good results.

For sure, I use same settings!
But I remove also all sequence parts and add the calculation, which I get from Erik in a forum. Guess there must be something wrong.

It would be very nice if you could upload a version, without GUI and History.

Thank you!

erikmansson, 2019-10-21:
The added calculations was for `sparse_fft.py`. `sparse_speed.py` does not use the same technique, so it won't work there. However, the steps to remove the plotting is the same. Basically, remove the stuff related to PyQtGraph (pg) in the `main()` function, like

https://github.com/acconeer/acconeer-python-exploration/blob/20a0a7638e2ed493c079ca9946a41002804907d0/examples/processing/sparse_speed.py#L41

You will get the speed output here:

https://github.com/acconeer/acconeer-python-exploration/blob/20a0a7638e2ed493c079ca9946a41002804907d0/examples/processing/sparse_speed.py#L54

erikmansson, 2019-10-26:
I'll close this for now. Please feel free to reopen this issue if you get stuck!

closed 2019-10-26 by erikmansson