acconeer/acconeer-python-exploration #14: Connect over socket

fohnbit, 2019-10-22:
Hello,

my XB112 is connected via USB on my rasbperry.

Is it possible to connect from my local windows machine to the sensor over a socket?

I use for other projects often ser2net ... but can I define the port in the gui (main.py).

Thank you!

erikmansson, 2019-10-22:
It is not possible to do so with the socket client because we use a different (JSON-based) protocol for that. I guess you would have to emulate a serial port on your local machine and then use regular UART client. Alternatively, you could look into swapping out the link used in the UART client. The link is instantiated here:
https://github.com/acconeer/acconeer-python-exploration/blob/d8291935f465ffb54ae9a3e3baf7978ab7cd8b52/lib/acconeer_utils/clients/reg/client.py#L30-L33

Replace that by the SocketLink:
https://github.com/acconeer/acconeer-python-exploration/blob/d8291935f465ffb54ae9a3e3baf7978ab7cd8b52/lib/acconeer_utils/clients/links.py#L50

fohnbit, 2019-10-25:
Hi Erik,

you mean I should replace "SerialLink" with "SocketLink"?

The ser2net made the UART as RAW over a TCP connection available. So it should be the same as I connect to a local UART ... but just over IP.

erikmansson, 2019-10-26:
Yeah exactly. Hopefully that'll work but you might need to modify SocketLink (for example the port). Please share if it works!

fohnbit, 2019-10-26:
I can test it on monday. Port I can keep the default in your code

fohnbit, 2019-10-28:
Hello Erik,

it is not working. Maybe you see the problem?

I setup ser2net on the PI with:
`6110:raw:0:/dev/ttyUSB0:115200 1STOPBIT 8DATABITS NONE`
Port: 6110, Baud: 115200 8N1

When I start a TCP connect, I get continued data:
`00 CD 7A 38 CD 2A 07 48 45 4F 08 08 65 4B 4C 6E 68 67 21 38 FD 63 45 6A 01 4F DD 69 A7 6F 43 55 97 08 E6 D4 61 67 41 7C F8 E2 41 D9 7F 42 08 5A 4D 27 38 1A 61 7A FF 1A 47 7D 0A 4D 61 68 65 AA 08 F2 91 42 FF DC B8 4B 63 49 07 25 67 1A 61 5C 3A 67 47 45 EF 2A 05 FC 21 65 67 7F FD C8 09 49 15 1A 5F 38 47 09 5B 6B 4C 6B DD 41 CC 6B 3A 45 47 7C 05 49 62 09 07 DC CE 7F 28 43 68 61 5C 61 05 6C CF 68 45 01 61 27 38 5E 41 63 C8 6F 78 58 28 67 6D 2A 49 DD CC 09 07 67 4C 03 03 47 67 4B 25 21 65 3A CD 0D 78 CB 28 BA 63 47 F6 4B 25 6A 3A 3A 21 5F E7 0A EE 07 49 01 29 ED 4D 4B 2A 41 EF DE 45 69 6D 67 4F 0F 5A F9 01 CD 25 58 61 21 38 67 67 DF 21 3A 43 7A 65 4D 29 23 07 FD 78 CD 68 6F EE 03 07 2A 0D 61 5A 6A 05 4D 0A 2A 61 01 6B 25 41 E3 6F 05 2A 25 5C B8 80 BA 21 21 67 BE DE CF 65 67 23 07 45 E5 43 23 F0 10 0B 2A 08 58 47 61 45 08 EE DF 0B 63 4F 4C 61 08 DA 27 41 61 63 F1 63 25 63 43 47 EE 65 78 3F 5A 4D 5E 47 45 78 67 E5 28 63 5D CA 2B 09 45 67 7C 03 EE 05 D9 41 49 4D 4C 38 03 CC 0D 21 CC 47 03 81 68 25 3A F8 58 2A 47 6E 08 05 C9 49 25 65 DF 5F 6B 5D 43 EF 99 25 D4 5F 45 78 7E 68 DF 29 FB 41 09 68 08 2D 3A DC 43 2F CE 4F 1A 05 61 CC DE 05 C9 7E 58 7A 03 47 78 5B 43 47 03 7D 5A 67 EF F8 DF 5A 6C 63 FE 63 2A 63 3A 67 4B CC 4D 79 7D FC C9 63 6D 6B 2D 4B 4D 6A DD 65 1A 4B 81 4D 38 EF 67 7B 68 CF 65 27 18 4A 9F 0D 29 28 2B C9 65 CD 7B 28 07 41 05 18 4E CC 45 18 1D CF 4F EE DF 67 CA 65 1F 41 7A 6A 4B CA FE 42 45 29 7E 08 CC 43 49 45 3A DF 2B EC 43 77 CE 07 43 29 CD 38 41 07 07 CB 41 CF 5D C5 5F 7A 28 25 D3 69 78 BA 61 EF 18 21 58 67 78 49 49 45 65 18 01 18 63 ED B7 98 25 78 08 5E 69 43 FC 45 65 EF 67 6B 6D 45 63 7B 29 6F 01 9F 63 45 18 ED 5E 45 43 68 78 DA DA CE 4B 67 27 01 38 4F 38 C`

Is this ok? The sensor sending data, without I send anything.

When I start the python code, the methode send 3 times:
`CC 05 00 F9 03 00 00 00 00 CD`
Then I get the error:
```
2019-10-28 11:07:59,607 - root - INFO - radarCat starting with args Namespace(debug=False, quiet=False, sensors=[1], serial_port='192.168.0.216', socket_addr=None, spi=False, verbose=True)
Itentify Windows IP address instead of a Com Port
Traceback (most recent call last):
  File "radarCat.py", line 489, in <module>
    main()
  File "radarCat.py", line 111, in main
    session_info = client.setup_session(sensor_config)
  File "C:\Users\shops\AppData\Roaming\Python\Python37\site-packages\acconeer_utils\clients\base.py", line 47, in setup_session
    self.connect()
  File "C:\Users\shops\AppData\Roaming\Python\Python37\site-packages\acconeer_utils\clients\base.py", line 24, in connect
    info = self._connect()
  File "C:\Users\shops\AppData\Roaming\Python\Python37\site-packages\acconeer_utils\clients\reg\client.py", line 87, in _connect
    raise ClientError("could not connect, no response")
acconeer_utils.clients.base.ClientError: could not connect, no response
```

So the sensor is available over TCP and sending data, but I get no response.

I disabled the check baudrate and handshake, but then I get the error:
`acconeer_utils.clients.base.ClientError: got invalid frame (incorrect start marker)`

I think the sensor should not send data by his own, right?

erikmansson, 2019-10-28:
That doesn't seem quite right. The server should be quiet if no session is
running. Can you make sure the buffer is empty before starting? Also, you
might run into baudrate issues. It starts at 115200 but is switched to 3M
upon a successful handshake at 115200. You might need to modify that. Run
the script with the flag -vv to get more info.

Sorry I can't help you more right now - I am and will be quite busy this
week.

fohnbit, 2019-10-29:
Hello Erik,

no problem. I made some tests. But:
How can I flush the buffer?

erikmansson, 2019-10-29:
You can do it by calling [socket.recv(bufsize)](https://docs.python.org/3/library/socket.html#socket.socket.recv) until it's empty.

The socket object is accessible through `client._link._sock`, so you would do something like
```
client.connect()
# time.sleep(1)  # might be needed
while client._link._sock.recv(4096):
    pass
```

erikmansson, 2019-11-15:
> Hello Erik,
> 
> no problem. I made some tests. But:
> How can I flush the buffer?

Guessing you accidentally reposted this.

Closing this. Feel free to reopen!

closed 2019-11-15 by erikmansson

battosai30, 2020-07-22:
Hi,

@fohnbit did you success ?

```
I'm trying to do the same thing but got this in exploration tool : 
Traceback (most recent call last):
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\links.py", line 99, in recv_until
    r = self._sock.recv(self._CHUNK_SIZE)
socket.timeout: timed out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\json\client.py", line 43, in _connect
    header, _ = self._recv_frame()
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\json\client.py", line 175, in _recv_frame
    packed = self._link.recv_until(b'\n')
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\links.py", line 101, in recv_until
    raise LinkError from e
acconeer.exptool.clients.links.LinkError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "main.py", line 1495, in connect_to_server
    info = self.client.connect()
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\base.py", line 30, in connect
    info = self._connect()
  File "C:\Users\Creatronic\Downloads\portable_exploration_tool\tools\python-3.7.3-embed-amd64\lib\site-packages\acconeer_exptool-3.6.3-py3.7.egg\acconeer\exptool\clients\json\client.py", line 45, in _connect
    raise ClientError("no response from server") from e
acconeer.exptool.clients.base.ClientError: no response from server
```

So XM122 kit is connected to a RockiPiS running ser2net (with the same config as fohnbit) and I'm trying to run exploration tool on Windows10 after modifications you suggested (and re-running python -m pip install -U --user . after modification as suggested in Git readme).

Regards