acconeer/acconeer-python-exploration #15: Plot of data against time

lucassbroto, 2019-10-22:
Good day!

In my project I am trying to get the phase shift data to track the movements of the chest for respiratory and heart rate. 

I know that the GUI already does a fine job with the respiratory rate. But I need the raw data to do my own processing and I am encountering a few problems. Inicially I would like to take the phase shift [rad] plotted against the time, but all the graphs are plotted agains depth and the one that is plotted against time (distance x time) is not the one of interest.

Another problem that I have been facing is regarding the performance of FFT in the graphs. Every time I try the exploration tool shuts down and I get the following message:
[image]

Any idea about what is happening or how I should procede to get the information that I need? Maybe the information displayed at the IQ  at peak graph, but in a linear display.

erikmansson, 2019-10-23:
Are you looking at _Breating_ or _Phase tracking_? The two obtain the phase delta in slightly different ways, but should have similar results.

The phase x time is basically the same as distance x time - there's just a factor ((2.5mm) / (2 pi rad)) between them. In _breathing_, that translation happens here:
https://github.com/acconeer/acconeer-python-exploration/blob/ce7faa9af7890145fb340f7aa734954153ac9463/examples/processing/breathing.py#L144-L145

In _phase tracking_, it happens here:
https://github.com/acconeer/acconeer-python-exploration/blob/ce7faa9af7890145fb340f7aa734954153ac9463/examples/processing/phase_tracking.py#L103

For the other problem I can't seem to recognize where it's crashing. What are you trying to do when it crashes?

erikmansson, 2019-10-24:
Sounds reasonable to start from the phase tracking example! Feel free to ask questions if you get stuck or anything is unclear.

We don't support PyQtGraph:s built in processing features. In fact, I didn't even know you could do that. :) So I think you'll have to do it on your own. Perhaps, you could take inspiration from for example the `sparse_fft.py` example.

erikmansson, 2019-11-02:
I'll close this now. Feel free to reopen this issue or create a new one if you want.

closed 2019-11-02 by erikmansson