acconeer/acconeer-python-exploration #16: Pause client streaming

fohnbit, 2019-10-27:
Hello,

is it possible to pause the streaming and later to resume?
for example:
`client.pause_streaming()`
`client.resume_streaming()`

Because when I just put the code to sleep 10sec, the server still send data and it seems he is buffering the data.

I want to pause the method, because need free performance on my Pi and after done all tasks, the method should continue streaming.

https://github.com/acconeer/acconeer-python-exploration/blob/15017e92dd607cfe1c656e59efb704b1734df234/examples/processing/sparse_speed.py#L52-L60

erikmansson, 2019-10-27:
Well, yes and no. In our C SDK, this would correspond to _activating_ and _deactivating_ the service.   Currently, this is only partly supported in the streaming and module servers, and we (currently) have no intention of adding this functionality. However, since you want to pause for such a long time, I think it will be no problem for you to simply use `start_streaming(config)` and `stop_streaming()` instead.

fohnbit, 2019-10-27:
Hi Erik,

I try this. Thanky ou!

closed 2019-10-27 by fohnbit

fohnbit, 2019-10-28:
Hi Erik,

I try this. Thank you!