acconeer/acconeer-python-exploration #17: Loading hdf5 files

lucassbroto, 2019-10-30:
Good afternoon,

How can I open a file generated by the gui in another environment, so I can use it?

The files are stored in Hdf5 format, everytime I try to open the file in python i get the following error.
[image]

Or an error saying that it cannot decode bytes ( I have tried using the raw string and the double \\, but so far it hasn't worked)

When I open the hdf5 in matlab, the value that I get is a single value for the whole file, which doesn't make sense to me.

[image]

Could you explain how can I access the data that the gui is interpreting or where can I find information on the way the data is saved?

dj-acconeer, 2019-10-31:
When you call the H5F.open function, it returns a file identifier.
That is the expected behavior.
After that you will need to read the data from the different fields in the h5 files.

You can read about how to do that with matlab on the mathworks site:
For instance
[https://www.mathworks.com/help/matlab/import_export/importing-hierarchical-data-format-hdf5-files.html#bsxixpv-2](url)

RilleN, 2019-10-31:
I was able to read some IQ-data with the MATLAB-code below. It might help.

filename = 'someIQdata.h5';
service_type = h5read(filename,'/service_type');
if strcmp(service_type{1}, 'iq')
    IQ_data = squeeze(    h5read(filename,'/real') ...
                     + 1i*h5read(filename,'/imag'));
end

session_info = h5read(filename,'/session_info');
session_info_struct = jsondecode(session_info{1});

r = session_info_struct.('actual_range_start') + ...
    linspace(0, session_info_struct.('actual_range_length'), ...
             session_info_struct.('data_length'));

plot(r,abs(IQ_data))
xlabel('x [m]')
ylabel('|IQ-data|')

dj-acconeer, 2019-11-05:
Closing this issue.
Please let us know if you still have issues.

closed 2019-11-05 by dj-acconeer

lucassbroto, 2019-11-05:
Hello, 

I've been playing around a bit with this code and with the structure of the file since your last suggestion, I appreciate you taking the time to clarify my questions.

The plot of the IQ data that you sent is regarding the absolute IQ value x distance, but my interest is in the phase information x time. I know how to get the phase angle but i can't find the time stamp in the datasets inside the H5.

In previous radar I have worked with, one of the groups inside the file, contained the time information, the timestamp. But this is how my hdf5 is structured. 
[image]

I need to syncronize my radar with another device, so this information is very important for me.

Thank You.

RilleN, 2019-11-05:
No timestamp is currently saved, but as long as the sequence_number increase by one for every sweep, the time between each sweep is 1/sweep_rate with high precision.

lucassbroto, 2019-11-05:
Ah, great then. Thank you so much!