acconeer/acconeer-python-exploration #18: presence_detection_sparse.py can't stream multiple sensor with EVK

GoldenRed, 2019-11-07:
I have the Raspberry Pi + EVK + 4 sensors and I want to stream data from all 4 sensors at once by running ``` python presence_detection_sparse.py -s 192.168.XXX.XXX --sensor 1 2 3 4 ``` . This works with the sparse.py service, it plots them all under each other, but doing it with presence_detection_sparse.py will have it saying that it does not recognize "2 3 4".

Is it like this on purpose or is it just an unintentional bug?

dj-acconeer, 2019-11-08:
Hi GoldenRed,
we are sorry for the inconvenience, but not all detectors are multi-sensor capable at this point.
We are working on adding this functionality to all detectors.

If you want to work with multiple sensors for presence detection at this point, you would need to write a wrapper around the presence detector.

GoldenRed, 2019-11-08:
I modified the file myself. 

https://github.com/GoldenRed/acconeer-python-exploration/blob/master/examples/processing/presence_detection_sparse_multi.py

closed 2019-11-28 by dj-acconeer