acconeer/acconeer-python-exploration #19: Radar Parameters

lucassbroto, 2019-11-28:
Hello, 

The doubt I have now is pretty simple but i believe it will be helpful for tha the radar beginners. I am using the IQ mode of measurement for further processing and extraction of information.

My methods of extraction of info is bu FFT and wavelets. However to proceed with my experiments I am still looking to find the best setting of the radar.

Reading through the documentation It is not clear a few things regarding the particular works of the radar such:
- Are the sample mode A and B equivalent to streaming and max frequency?
- What is the meaning of the HW Acc avg samples? Is it regarding the pulse integration?
- What does the step size, image history and averaging influence in?
- The update rate is equivalent to the aquisition frequency of the radar? 
- Is there a way to perform my measurement without skipping frames?

Thank you 
Lucas Broto.

erikmansson, 2019-11-28:
> Are the sample mode A and B equivalent to streaming and max frequency?

No. The sampling mode changes how averaging is done in the sensor. Mode A is optimized for reducing correlation between samples. Mode B is optimized for SNR, and works better with higher values of HWAAS.

> What is the meaning of the HW Acc avg samples? Is it regarding the pulse integration?

Yes. HWAAS sets how many pulse samples that are averaged together for every point in the buffer. Allowed values are 1 - 63 inclusive. _Ideally_, SNR grows linearly as you increase HWAAS.

> What does the step size, image history and averaging influence in?

**Stepsize:** Sets the distance between sampled points. A stepsize of 1 corresponds to 0.5 mm which is the densest possible. 2 means 2 x 0.5 = 1 mm, 3 means 3 x 0.5 = 1.5 mm, and so on. The IQ and Envelope services support setting stepsize to 1, 2 or 4.

**Image history:** Just sets the number of sweeps shown in the plots

**Averaging:** Sets the [smoothing factor](https://en.wikipedia.org/wiki/Exponential_smoothing) for filtering the sweeps over time. 0 corresponds to no smoothing. 0.99... means maximal smoothing.

>The update rate is equivalent to the aquisition frequency of the radar?

Can you elaborate?

>Is there a way to perform my measurement without skipping frames?

I guess you want to measure as fast as possible? In the next release, this will be very easy. In the current release, you can enable "experimental stitching" and set the update rate to a very high value, like 1000.

We will soon release improved documentation which hopefully gives you a better idea of how all these parameters work.

lucassbroto, 2019-11-28:
Thank you for your fast answer.

I have done some calculations regarding my number of points and distance measured and did get 0,5mm.

All the information that you provided me was very very helpful.

Regarding your question about the acquisition frequency: 
The sweep rate sets the time between taken points in the signal, so I am assuming that it is the fast time acquisition frequency - regarding the sampled points that constitute the range line, right? But when I set it as '30', does it mean 30 Hz? And what is the top limit?

Another information is the PRF and bandwidth, how can I set it or access its values? This is related to the information about the "experimental stitching" as well.

Thank you!

erikmansson, 2019-11-29:
I think you might get your question answered by the sensor introduction document, found [here](https://developer.acconeer.com/download/introduction-to-acconeers-sensor-technology-pdf/). See section 3. Please get back to us if that doesn't clear things up. We're working on improving this documentation as we speak.

The PRF is fixed to 13 MHz and the pulse shape is set by the profile. "Experimental stitching" does not affect how the radar operates. The coming release will give you more freedom of setting the profiles, so sit tight :)

lucassbroto, 2019-12-02:
Thank you so much for you answer. Is there a prevision for the next release?

erikmansson, 2019-12-03:
Currently, no, but it will be out soon!

closed 2019-12-03 by erikmansson

erikmansson, 2019-12-06:
It's out :)