acconeer/acconeer-python-exploration #20: Trying to use XC112/XR112 with NUCLEO-F401RE

chris1295, 2019-12-02:
Hello :)  I'm trying to use the evaluation kit with a Nucleo-F401RE. I've got a problem with the SPI communication, then I saw in the code provided to implement this kit on STM32 and there is a file called "acc_hal_integration_single_thread_stm32cube_xc112.c". In this file, there is a function to power on the sensor, but I'm not sure if I've declared all the required pins. To be especific, I'm a little bit confused about the settings for the slave selector.
This is the function:
_static void acc_hal_integration_sensor_power_on(acc_sensor_id_t sensor_id)
{
...
HAL_GPIO_WritePin(**A111_SPI_SS**_GPIO_Port, A111_SPI_SS_Pin, GPIO_PIN_SET);
...
	// Select communication with only Sensor 1 and deselect all other radar slots
HAL_GPIO_WritePin(**XC112_SPI_S1_ENABLE**_N_GPIO_Port, XC112_SPI_S1_ENABLE_N_Pin, GPIO_PIN_RESET);
}_
I believed that I needed only the *XC112_SPI_S1* pin to communicate with the sensor but the **A111_SPI_SS** pin confuse me a bit. Actually, I cannot see the last one in the electrical schematic . I guess the **XC112_SPI_S1** pin is the pin called **SPI_S1_ENABLE_N** in the schematic but the other one, I'm not sure where it is.
[image]

vackraetraed, 2019-12-03:
Hello @chris1295 ,

Anton from Acconeer here. 
I believe A111_SPI_SS is the signal called SS_N in the schematics.
SS_N works like normal SS/CS.

Setting HAL_GPIO_WritePin(A111_SPI_SS_GPIO_Port, A111_SPI_SS_Pin, GPIO_PIN_SET);
means that you are not communicating SPI at the moment.
This signal needs to be set low when transferring and set high after spi transfer is done.

HAL_GPIO_WritePin(XC112_SPI_S1_ENABLE_N_GPIO_Port, XC112_SPI_S1_ENABLE_N_Pin, GPIO_PIN_RESET).
means that SPI communication will only be done to sensor 1.

chris1295, 2019-12-03:
Thank you! Now, it works correctly.

closed 2019-12-04 by chris1295