acconeer/acconeer-python-exploration #21: Error with displaying GUI (couldn't connect to MK112 board via serial communication)

msallmen23, 2019-12-10:
I keep getting the following message when trying to run GUI. Please view the picture attached. 

Let me know if anyone has any ideas on how to fix this.

I have tried the following:
- installed python 3.6
- uninstalled all other versions of python
- used PuTTy to verify USB port knows MK112 board is connected
- Tried to make virtual environment for python using visual studio. (I still get the same error)
Please follow this Link:  https://ezgif.com/image-to-datauri/ezgif-3-a90847167c7a.png 

Picture:  [image]

dj-acconeer, 2019-12-10:
Could you verify that COM3 disappears when you unplug the module?

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
It's not showing up. Or where you referring to COM3 showing up on the GUI?

dj-acconeer, 2019-12-10:
In order to help you best, I need to understand your setup.
First step is to make sure that the USB module is actually on COM3.
Are you running this in a VM or directly in Windows?

msallmen23, 2019-12-10:
Directly on Windows. It is on COM3. 
[image]

msallmen23, 2019-12-10:
Thanks for the assistance thus far.

dj-acconeer, 2019-12-10:
From your other screenshot I can see that you are using the latest exploration tool with API 2.0.
We just released this new API version last week and there might be incompatibility issues.
When did you flash the software for the usb module?

msallmen23, 2019-12-10:
Last week. Should I install previous version? If so, can you provide link to previous version that you recommend for my current enviornment.

dj-acconeer, 2019-12-10:
I think first we could try to flash the usb module gain to make sure it's the latest.

msallmen23, 2019-12-10:
What do you mean by flash the usb module?? I apologize I am a newbie to RF.

dj-acconeer, 2019-12-10:
No problem. We will guide you through it.
Follow the instructions here to flash the latest firmware to the USB module

https://acconeer-python-exploration.readthedocs.io/en/latest/evk_setup/xm112.html

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
I ran the first script in the link you provided and that was the previous screenshot

dj-acconeer, 2019-12-10:
That means, the srcipt found a device attache to com3.
If you could download bossa and the latest firmware and flash that to your module, we can make sure you have the latest firmware on the module.

msallmen23, 2019-12-10:
ok, I am not able to download the link provided...

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
the module sw user guide doesn't upload for me

msallmen23, 2019-12-10:
I am on the Acconeer Developer site and am looking for that same "module software user guide". Can you point me in the right direction from the website?

msallmen23, 2019-12-10:
Thanks again for all your help. I really appreciate it.

dj-acconeer, 2019-12-10:
I looks like the link is outdated.
You need to go to our homepage at
https://developer.acconeer.com
Navigate to the "Developer" section and create an account.
Then login here
https://developer.acconeer.com/log-in/
and download the latest software.

msallmen23, 2019-12-10:
Is this the right one?
[image]

dj-acconeer, 2019-12-10:
Yes.

msallmen23, 2019-12-10:
Ok, great.

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
THat is the new error that I am getting.

msallmen23, 2019-12-10:
Should I try this as well? 
[image]

msallmen23, 2019-12-10:
Nevermind, it is letting me write now. I had to perform a read first. Now it's letting me write.  
[image]

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
This is the error that I am getting now.

dj-acconeer, 2019-12-10:
In your screenshot, it looks like you flashed the bossac.exe to the USB module.
Make sure you select the firmware image you downloaded.
When you have flashed, you might have to unplug and plug back in the module.

msallmen23, 2019-12-10:
[image]

msallmen23, 2019-12-10:
Ok, that was all that I installed from BOSSA. I will check again. THanks again for the feedback. I got the Bossa GUI to write successfully, however I will verify it is right firmware. I will take sc of that.

msallmen23, 2019-12-10:
still getting the same error.

dj-acconeer, 2019-12-10:
What is the name of the file that you flashed?

msallmen23, 2019-12-10:
bossa-dpinst.wxs

msallmen23, 2019-12-10:
It said it was windows driver signature in github repository. However I am sure that is not the right file. I tried searching for a .bin file in the Bossa 1.9.1 and couldn't find one.

msallmen23, 2019-12-10:
[image]

dj-acconeer, 2019-12-10:
Please  download this file

acconeer_xm112_module_software_v2_0_0.zip

on the developer page labeled as

acconeer_xm112_module_software

From this zip file, please extract the binary

acc_module_server_xm112.bin

Then flash that file to your module.

msallmen23, 2019-12-10:
Ok, thank you. I will try this.

msallmen23, 2019-12-10:
I did precisely that and I am still getting the same issue.  
[image]

dj-acconeer, 2019-12-10:
[image]

It should look something like this.
And after flashing, unplug the module and plug it back in.

dj-acconeer, 2019-12-19:
Do you still have issues with this?

dj-acconeer, 2019-12-20:
Closing this for now

closed 2019-12-20 by dj-acconeer