acconeer/acconeer-python-exploration #22: Stability problems with Version 2.0.0

GregorBader, 2019-12-10:
Using XM122 EVK with Portable Exploration Tool on native Windows:
```
10:51:26.668 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:51:26.675 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:51:33.741 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:51:41.806 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:51:43.071 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | data length: 0
10:51:43.072 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | buffer size: 0 B
10:51:43.073 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | requested data rate: 0.00 Mbit/s
10:51:43.073 | WARNING | MainProcess      | acconeer.exptool.clients.base | actual measured range differs from the requested
Traceback (most recent call last):
  File "main.py", line 1782, in plot_timer_fun
    self.service_widget.update(data)
  File "D:\work\Acconeer\portable_exploration_tool\acconeer-python-exploration\examples\processing\phase_tracking.py", line 190, in update
    com_x = (1 - data["com"]) * self.depths[0] + data["com"] * self.depths[-1]
IndexError: index 0 is out of bounds for axis 0 with size 0

10:52:27.220 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:52:27.223 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:52:41.427 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:52:54.359 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:52:55.615 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | data length: 620
10:52:55.616 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | buffer size: 2480 B
10:52:55.616 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | requested data rate: 0.40 Mbit/s
10:54:10.001 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:54:16.494 | INFO    | MainProcess      | acconeer.exptool.clients.base | reported version: 2.0.0
10:54:17.729 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | data length: 620
10:54:17.730 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | buffer size: 2480 B
10:54:17.730 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | requested data rate: 0.60 Mbit/s
10:54:17.731 | WARNING | MainProcess      | acconeer.exptool.clients.base | actual measured range differs from the requested
Traceback (most recent call last):
  File "main.py", line 2028, in run
    _, record = self.radar.process(sweep, info)
  File "D:\work\Acconeer\portable_exploration_tool\acconeer-python-exploration\gui\data_processing.py", line 93, in process
    out_data = self.external.process(in_data)
  File "D:\work\Acconeer\portable_exploration_tool\acconeer-python-exploration\gui\service_modules\iq.py", line 85, in process
    self.history[-1] = data
ValueError: could not broadcast input array from shape (1,1240) into shape (1,620)
```

erikmansson, 2019-12-10:
I've tried reproducing the error but couldn't. To get as close as possible to yours, I tested with
- Windows in VirtualBox
- Exploration commit 2469c1d
- XM122 flashed with v2.0.0

Some questions:
- Are you running v2.0.0 on the XM122?
- Did you change any settings?
- Did you upgrade Exploration from v1 or clean setup directly to v2?

GregorBader, 2019-12-10:
Yes, running v2.0.0 on XM122. It happens with fresh setup of exploration tool without any changed settings.

erikmansson, 2019-12-10:
Ok. Does other examples/modules run? Like IQ or breathing (IQ)?

GregorBader, 2019-12-10:
The errors are not persistent!
When they occur, a USB disconnect resets the port, and the same example might run afterwards without problems.
It seems to me, that it is timing related when data arrives from XM122.

GregorBader, 2019-12-10:
@erikmansson I'm now able to reproduce the error:
```
Traceback (most recent call last):
  File "main.py", line 2028, in run
    _, record = self.radar.process(sweep, info)
  File "D:\work\Acconeer\portable_exploration_tool\acconeer-python-exploration\gui\data_processing.py", line 93, in process
    out_data = self.external.process(in_data)
  File "D:\work\Acconeer\portable_exploration_tool\acconeer-python-exploration\gui\service_modules\envelope.py", line 124, in process
    self.bg_buffer[self.data_index] = data
ValueError: could not broadcast input array from shape (1,1281) into shape (1,661)
```
everytime I change the range interval. When I start the measurement, the envelope graph scales change to the new range interval, then change back to previous value and when the data should be displayed the error occurs.

erikmansson, 2019-12-10:
Ok thanks. I'll look into it and get back to you.

erikmansson, 2019-12-14:
I noticed now that you got this log entry:

> 10:51:43.071 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client | data length: 0

This seems odd to me. Can you reproduce it? Or can you perhaps remember what you were doing when this appeared?

GregorBader, 2019-12-16:
Yes, I can reproduce it. It happens every time after a Power On of the XM122 EVK.

erikmansson, 2019-12-16:
I've tried again to reproduce it but didn't succeed. Could you try replacing this line
https://github.com/acconeer/acconeer-python-exploration/blob/24717c501b5f2ccb7e7d02562b29c404f24f4a47/src/acconeer/exptool/clients/reg/client.py#L82
with `sleep(1)`?

Don't forget to re-run the install after the modification:
`pip install -U --user .`

GregorBader, 2019-12-17:
Hi @erikmansson ,

The `sleep(1)` did not help, but a `sleep(5)` did.

GregorBader, 2019-12-17:
I would suggest increasing `_STATUS_TIMEOUT` in line https://github.com/acconeer/acconeer-python-exploration/blob/24717c501b5f2ccb7e7d02562b29c404f24f4a47/src/acconeer/exptool/clients/reg/client.py#L45

and raising an error in `_wait_status()` when timed out.

erikmansson, 2019-12-17:
Oh good catch, that was an oversight on my part. Of course there should be an error. Pushing a fix soon. Thanks!

erikmansson, 2019-12-17:
Fixed in cfa43b8. Hopefully this works! I increased the timeout to 3 seconds. Should be enough.

GregorBader, 2019-12-17:
Verified

closed 2019-12-17 by GregorBader

erikmansson, 2019-12-17:
Perfect! Thanks :)