acconeer/acconeer-python-exploration #23: XM122 GUI getting server communication error

Autobot86, 2019-12-12:
Hello I got issue on XM122 module interface with my laptop
i followed all steps but when start measurement click getting this error 

[image]

erikmansson, 2019-12-12:
What version (commit) are you using?

Autobot86, 2019-12-12:
which version EVM or DFU...??

erikmansson, 2019-12-12:
I was thinking about the Exploration Tool itself, but module SW version is also relevant. It should be v2.0.0.

Autobot86, 2019-12-12:
yes SW version is v2.0. Exploration Tool version is V3.0.7.

erikmansson, 2019-12-12:
Alright, should work. The current version of Exploration gives more detailed error messages if you click "Show Details...". Could you upgrade, reproduce the error, then supply us with the traceback given in the details?

Autobot86, 2019-12-13:
I already upgrate the new version its v3.0.7 is there any new update available..?? 

C:\Users\PC\Downloads\acconeer\acconeer-python>python gui/main.py
10:02:47.897 | INFO    | MainProcess      | acconeer.exptool.clients.base
 | reported version: 1.10.0
10:02:47.898 | WARNING | MainProcess      | acconeer.exptool.clients.base
 | old server version - please upgrade server
10:02:47.933 | INFO    | MainProcess      | acconeer.exptool.clients.base
 | reported version: 2.0.0
10:02:53.571 | INFO    | MainProcess      | acconeer.exptool.clients.base
 | reported version: 2.0.0
10:02:55.574 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client
 | data length: 1240
10:02:55.575 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client
 | buffer size: 2480 B
10:02:55.576 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client
 | requested data rate: 0.60 Mbit/s
Traceback (most recent call last):
  File "gui/main.py", line 2054, in run
    info, sweep = self.client.get_next()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\a
cconeer\exptool\clients\base.py", line 83, in get_next
    return self._get_next()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\a
cconeer\exptool\clients\reg\client.py", line 251, in _get_next
    reg = regmap.get_reg(addr, self._mode)
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\a
cconeer\exptool\clients\reg\regmap.py", line 188, in get_reg
    raise ValueError("unknown reg: {}".format(value))
ValueError: unknown reg: 159

This errors are provide in command prompt

erikmansson, 2019-12-13:
Thanks for the traceback. Found a bug which I fixed in 6dc02f1.

However, this _shouldn't_ happen with SW v2.0.0. Can you double check that it's actually running this version? If so, we might have another bug.

Autobot86, 2019-12-13:
I reinstall the software it's given same error again and again

erikmansson, 2019-12-13:
Ok, we'll look into it! Did my fix solve your problem?

Autobot86, 2019-12-16:
Still i have same issue it doesn't work same this kind of error provide

File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\a
cconeer\exptool\clients\reg\client.py", line 251, in _get_next
reg = regmap.get_reg(addr, self._mode)
File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\a
cconeer\exptool\clients\reg\regmap.py", line 188, in get_reg
raise ValueError("unknown reg: {}".format(value))
ValueError: unknown reg: 159

erikmansson, 2019-12-16:
Did you re-run `pip install .` after updating?

Autobot86, 2019-12-18:
I re-install pip now i get another error of COM port Traceback as below

Traceback (most recent call last):
  File "gui/main.py", line 1435, in connect_to_server
    info = self.client.connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\base.py", line 24, in connect
    info = self._connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\reg\client.py", line 207, in _connect
    self._link.baudrate = product_max_baudrate
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\links.py", line 174, in baudrate
    self._ser.baudrate = new_baudrate
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\serial\serialutil.py", line 295, in baudrate
    self._reconfigure_port()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'The parameter is incorrect.', None, 87)

erikmansson, 2019-12-18:
Can you check so that the device is properly installed in the Windows device manager?

Autobot86, 2019-12-19:
Yes device installed my PC it show COM3 below image 

[image]

but Exploration also detect COM3. connection doen't make with PC to EVM

[image]

erikmansson, 2019-12-19:
Looks ok. Can you provide output from running this script?

    python examples/utils/ping.py -u COM3 -vv

Autobot86, 2019-12-19:
C:\Users\PC\Downloads\acconeer-python-exploration-master>python examples/utils/ping.py -u COM3 -vv
15:00:32.676 | DEBUG   | MainProcess      | acconeer.exptool.utils               | logging configured
15:00:32.698 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00
15:00:32.710 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | handshake succeeded at 115200 baud
15:00:32.711 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg r req: addr: 0x12
15:00:32.726 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | recv reg r res: addr: 0x12 val: 00 00 00 00
15:00:32.727 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | switching to 0 baud...
15:00:32.728 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x07 val: 00 00 00 00
15:00:32.742 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | recv reg w res: ok
Traceback (most recent call last):
  File "examples/utils/ping.py", line 28, in <module>
    main()
  File "examples/utils/ping.py", line 20, in main
    info = client.connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\base.py", line 24, in connect
    info = self._connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\reg\client.py", line 207, in _connect
    self._link.baudrate = product_max_baudrate
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\links.py", line 174, in baudrate
    self._ser.baudrate = new_baudrate
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\serial\serialutil.py", line 295, in baudrate
    self._reconfigure_port()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'The parameter is incorrect.', None, 87)

erikmansson, 2019-12-19:
Ok good. Now try changing this line:

https://github.com/acconeer/acconeer-python-exploration/blob/3f2f16018fe5daf7d41ca13df3be9d285b3d77e4/examples/utils/ping.py#L15

to

        client = UARTClient(port, override_baudrate=115200)

and re-run

Autobot86, 2019-12-19:
I changed that line still i got same COM port error

[image]

erikmansson, 2019-12-19:
It's unlikely that it is exactly the same. Can you send the terminal output from running it?

erikmansson, 2020-01-08:
Any progress?

erikmansson, 2020-01-09:
I'll close this for now. Feel free to reopen!

closed 2020-01-09 by erikmansson