acconeer/acconeer-python-exploration #26: xm112 distance measurement

cosmogenetic, 2020-01-08:
I want to measure the distance to the car using xm112 + xb112.
On the side of the car, only 2 meters are measured.
Only 1 meter is measured from the front of the vehicle.
The measurement was done using envelope service.
How can I measure up to 5 meters?
Please let me know the measurement method and option.

RilleN, 2020-01-08:
Hi cosmogenetic,

Can you specify what happen when measuring. Is software reporting an error or are the objects not detected? I have no problems collecting envelope sweeps from 0.2 to 6.8 meters using our latest software with xm/b112. If you inspect the envelope sweeps in the exploration tool, do you see any signs of the car?

To get good signals from objects at 5 meters, a lens is necessary. Also, it helps if the object is flat, like the side of a car, a wall or a fluid surface. To increase the signal, I recommend using a high profile, please try 3 or 4.

cosmogenetic, 2020-01-08:
Thank you RilleN.
No lens was used.
The measurement results were the same in the high profile.
Let's see how the results change after using the lens.
Are there any other options or methods?

RilleN, 2020-01-09:
Hi cosmogenetic,

For distance measurements with longer distances, the Envelope service with lens and a high profile is what we recommend. 

The presence detector have a very good SNR and can detect objects many meters away. However, the object has to move relative to the radar to be detected and the distance is not as accurate as for the Envelope.

/RIkard

cosmogenetic, 2020-01-14:
Thank you RilleN.
I used a lens and the measurement is good.
However, there is one more question.
I want to manufacture a product with A111. I will also make a lens.
However, we do not know what material the lens is made from, and there is no CAD drawing of the lens.
The video about LH112 says ABS plastic, but there are many ABS plastics.
Can you tell me the material of the lens?
Can I get a CAD drawing of the lens?

RilleN, 2020-01-14:
Hi cosmogenetic,

I think you will find relevant information in chapter 5.3 and 6 in our hardware integration guideline document: https://developer.acconeer.com/download/hardware-and-physical-integration-guideline-pdf/

Let me know if you have any further questions.

/Rikard

cosmogenetic, 2020-02-03:
Hi RilleN.
This time I'm going to test xr112
However, xc112 is not used.

I have a few questions before the test.

1. I'm trying to create a new board that doesn't use xc112 and raspberry pies for xr112
The new board wants to use the Atmel ATSAME70Q20A microcontroller.
Is there a better way?

2. Is it possible to measure the xr112 module up to 5 to 10 meters with a readome?

3. Please let me know your email address. There are some questions that I don't want to open for production and production. My e-mail is hyd.codeworks@gmail.com.

RilleN, 2020-02-04:
Hi cosmogenetic,

Regarding question 2, the software today allows measurements up to 7 meters. At this range it is possible to detect moving persons with the presence detector and the distance to a flat object, such as a fluid surface or wall, using the envelope service. This range do, however, require the use of a plastic lens or a horn antenna.

Regarding question 1 and 3 I suggest you post you questions on our freshdesk site: https://acconeer.freshdesk.com. Then your questions will be forwarded to the right competences within Acconeer.

cosmogenetic, 2020-02-19:
Hi RileN.
Thank you very much.
It helped a lot to make iot devices.

closed 2020-02-19 by cosmogenetic