acconeer/acconeer-python-exploration #27: Sensor_driven repetition_mode limits

stojanovicnikola68, 2020-01-10:
Hi,

I am playing around with obstacle_detection script. When I changed sensor_config.range_interval = [0.1, 1.0] I encountered following error when session_info = **client.setup_session**(sensor_config) is called:

15:03:30.899 | INFO    | MainProcess      | acconeer.exptool.clients.base        | reported version: 2.0.0
 File "/acconeer/exptool/clients/reg/client.py", line 75, in _setup_session
    self._wait_status(regmap.STATUS_FLAGS.CREATED)
  File "/acconeer/exptool/clients/reg/client.py", line 118, in _wait_status
    raise SessionSetupError
acconeer.exptool.clients.base.SessionSetupError

I noticed it can work in **SENSOR_DRIVEN** mode with different values of range_interval if I also change other settings like update_rate, downsampling_factor, hw_accelerated_average_samples, etc...

In **HOST_DRIVEN** I did not notice setup errors with any configuration.

So I assume in SENSOR_DRIVEN mode there must be some check weather the requested sample rate is achievable with desired settings. 
Am I right?
When is **STATUS_FLAGS.ERROR_CREATION** raised?

erikmansson, 2020-01-10:
> So I assume in SENSOR_DRIVEN mode there must be some check weather the requested sample rate is achievable with desired settings.

Yes, amongst other things. Here I think the issue here might be that the range is too large. Check if the error persists with a shorter range like [0.1, 0.5]. 0.9m is right on the limit of what the sensor can handle with downsampling_factor 1.

> When is STATUS_FLAGS.ERROR_CREATION raised?

When the sensor stack (running in the server) cannot verify the configuration or the calibration fails. In 99% of the cases, it's the former.

stojanovicnikola68, 2020-01-10:
Thank you for your response. Can you roughly say what are the range limits for different downsampling_factors?
It works with no problem with shorter range. Though it is happening only in **SENSOR_DRIVEN** mode. Does that mean that I could use wider range in **HOST_DRIVEN**?

erikmansson, 2020-01-11:
The limitation is due to the sensor buffer which holds up to 2048 points. The buffer usage for the different services is on average approximately:

- IQ:
  124 * (2.5 + `range_length` / 0.06) / `downsampling_factor`
- Envelope / Power bins:
  124 * (0.5 + `range_length` / 0.06) / `downsampling_factor`
- Sparse:
  `sweeps_per_frame` * (1 + `range_length` / 0.06) / `downsampling_factor`

This gives some approximate range limits:

- IQ: 0.84 m * `downsampling_factor`
- Envelope / Power bins: 0.96m * `downsampling_factor`
- Sparse: N/A in most cases

The buffer usage may vary depending on how the configured range aligns with the internal measurement grid. If you need to optimize this, please ask for our help!

In the `HOST_DRIVEN` mode*, multiple buffers can be *stitched* together which bypasses this limitation.

\* The sparse service inherently does not support stitching

erikmansson, 2020-01-20:
Closing this for now. Feel free to reopen if you have further questions.

closed 2020-01-20 by erikmansson