acconeer/acconeer-python-exploration #29: Max range

stojanovicnikola68, 2020-01-20:
Hi all,

I am using XB122 evaluation kit. I have noticed that A111 datasheet states: 

> Measures absolute range up to 2 m", 

while services allow range up to 7m and module description states up to 10m. Can you explain me what is the difference?

Also, I have some difficulty seeing a hedge/tree. I get no peaks in envelope for distances which are let's say roughly above 2m. I tried with different gains as well as with lens but the outcome was similar. Is this due to poor reflectivity of the objects or I am missing something?

Thanks!

dj-acconeer, 2020-01-20:
You are right about the reflectivity limiting maximum distance for visibility of objects.

The hardware supports measuring beyond 7m.
The software, by default, allows measuring up to 7m.
A lot of objects are visible up to 2m.

An objects' visibility depends on its material properties and its shape.
Untreated wood is a material with very low reflectivity when it's dry and might become undetectable at as little as 50 to 100cm.
Metal on the other hand is an extremely good reflector and can often be seen several meters away even without a lens.

Regarding shape, flat (or concave) objects perpendicular to the radar propagation direction can be seen better than convex or flat objects at oblique incidence.

If you can live with integrated measurements, you could try to use IQ data and ramp up the averaging.
Provided the hedge/tree or object you are trying to look at is stationary, this will coherently add the signal and improve SNR over time.

stojanovicnikola68, 2020-01-20:
Thank you @dj-acconeer for clarifying the behavior. 

> If you can live with integrated measurements, you could try to use IQ data and ramp up the averaging.
Provided the hedge/tree or object you are trying to look at is stationary, this will coherently add the signal and improve SNR over time.

Can you please explain a bit deeper what did you mean?
I tried obstacle_detection which is based on IQ, but I did not visually see anything useful for better tree obstacle detection.

Since I am trying to evaluate if I can use radar system based on a111 for drone obstacle detection, I am very interested in "tree response". 50 to 100cm would not be the enough range. From your experience what would be roughly the maximum detection distance for a tree?

dj-acconeer, 2020-01-22:
Hi @↕stojanovicnikola68, sorry for the delayed response.

If you want to use our sensor on a drone, you cannot rely on coherent integration of subsequent sweeps to improve SNR as your objects will not be stationary with respect to the drone (assuming your drone is moving).
This might prove a difficult use case, but i would recommend a different approach.
You could try using our sparse presence detection.
It is extremely sensitive and if you use profile 3 or 4, you should be able to get a good range response of several meters.
You should keep in mind though that for presence detection, object need to be moving relative to the sensor.
In your case, since the drone will be flying/moving, everything around it will be moving with respect to the sensor.
When you test sparse presence, please keep this in mind.
You will likely not see a try with spare presence if both, the sensor and the tree are not moving.

closed 2020-01-28 by dj-acconeer