acconeer/acconeer-python-exploration #30: RAW data

PAk-CatchFire, 2020-01-22:
Hello. 

We are working with your XM112 device. Since we wanted to use our own processing algorithms, we would like to know if it would be possible to extract Raw data from the device, even before FFT. Envelope and IQ Phase examples are a little bit cooked for us. 

We have used our functions with coherent FMCW systems, and we really think your system could benefit from those too, since the output we are seeing is similar in the Pulsed vs FMCW.

Kind Regards

erikmansson, 2020-01-23:
We currently do not support outputting raw data from the envelope, power bins, and IQ services, but are considering the possibility of supporting it in the future.

The C SDK supports disabling filtering in the depth dimension of IQ data, which might be of interest to you. Unfortunately, in v2.0.0 and the upcoming v2.1.0 release, this parameter is not settable in the module software. I will make sure that the release after v2.1.0 supports it.

Side note: The service data is not Fourier transformed, but demodulated.

PAk-CatchFire, 2020-01-23:
Thank you Eric.
Would you mind to point us to the file of the C SDK where we can disable the filters?
I assume it is the acc_service_iq_depth_lowpass_cutoff_ratio_set function, right?

Do you provide information a bout the process done inside the sensor (sampling, demodulation, filtering, gain, etc...) besides the Introduction to Acconeer's sensor technology.

How are you calculating the phase?

Regards

erikmansson, 2020-01-23:
Yes `acc_service_iq_depth_lowpass_cutoff_ratio_set` is correct. Setting it to 0.5 will disable the filter. This gives you IQ-demodulated raw data, and the phase is obtained from depth filtering it. I can not provide more information about the processing at this time.

PAk-CatchFire, 2020-01-23:
Thank you again Eric.
It would be great to see RAW data on 2.1.1 release!!  ;D Using Machine Learning could be really interesting with it.

I understand you provide the phase from the main target, but is it arctan(I/Q) channels directly, since you don't use FFT. 

On the other hand, could we disable the filter via acconeer-python-exploration? Is it possible to use it with the module software itself?

Sorry for all the questions, but we just landed on your hardware!!
Best regards

erikmansson, 2020-01-24:
> I understand you provide the phase from the main target, but is it arctan(I/Q) channels directly, since you don't use FFT.

Yes

> On the other hand, could we disable the filter via acconeer-python-exploration? Is it possible to use it with the module software itself?

In the C SDK it's already supported, but the module software (which extends upon the SDK) does not support it. Therefore it is currently not supported in Exploration.

PAk-CatchFire, 2020-01-24:
Thank you Eric.
Do you share the module software for compiling? Or at least, do you have the protocol of the module software?

erikmansson, 2020-01-24:
No, not at this time. The protocol is documented in the [user guide](https://developer.acconeer.com/download/xm112-module-software-user-guide-pdf/). You can also look at how the UART/SPI (reg) clients are implemented in this repository.

closed 2020-02-03 by erikmansson

PAk-CatchFire, 2020-02-07:
Hello Erik.

One question. What is the resolution of the distance for the Phase trakicng (IQ) example? Is it the same on the Breathing(IQ) example ?
What are the limits of the range on these examples? 
Sorry but I cannot find the information on the datasheet.

Thank you

erikmansson, 2020-02-07:
This is specified in the [datasheet](https://developer.acconeer.com/download/a111-datasheet-pdf/) under section 6.2. The average phase jitter (std dev) is 6.1 deg translating to 42 um in relative distance.

The range is heavily dependant on what kind target you are tracking. In the [radar sensor introduction](https://acconeer-python-exploration.readthedocs.io/en/latest/sensor_introduction.html#typical-ranges-for-different-objects) you can find typical ranges for common objects. However, the only way to know for sure is to test and measure.

PAk-CatchFire, 2020-02-07:
Thank you Eric.
So, if I compile and upload one of the C SDK examples to your  XM112, will we get communication with the exploration tool?
Do you have an ICD document  (or similar) for the exploration tool or the SDK?

Best Regards

erikmansson, 2020-02-07:
We use the module software (based on the SDK) for communication with PET. At this time, you can't compile that on your own.

There is a user guide [here](https://developer.acconeer.com/download/xm112-module-software-user-guide-pdf/) for the module software/protocol. Currently there are some minor parts which are out of date. Please feel free to ask questions about it!

PAk-CatchFire, 2020-02-08:
Thank you.
I have tried to repeat the 5.2 Reading Power Bin Data (UART Streaming) commands, with a Terminal program from Windows, but I do not see any response from the device.
Do you have any procedural test for the communication port?

erikmansson, 2020-02-09:
> The Read Status Register example does provide a response

Good!

> Is the sequence updated to 2.1 version?

Yes, the protocol changed in 2.0 and had a minor update in 2.1. I've had a similar question before, to which I replied:

How to start and stop services is not properly documented in the guide. For example, when sending a request to start the service (setting reg 0x03 (main_control) to 0x03 (create_and_activate)), you will get a response immediately. But this does not mean that the service has started - you have to poll the status register to check that. I think it's easiest to look at how the Exploration tool client does it. The status polling function is defined [here](https://github.com/acconeer/acconeer-python-exploration/blob/31a370237d6f17db3603f4d1c4bb43f9afda2f7c/src/acconeer/exptool/clients/reg/client.py#L102), and is used [here](https://github.com/acconeer/acconeer-python-exploration/blob/31a370237d6f17db3603f4d1c4bb43f9afda2f7c/src/acconeer/exptool/clients/reg/client.py#L76) when setting up a session. Note that Exploration first executes a create and then an activate, but [...] it's likely easier to use the combined command create_and_activate.

An up to date register map can be found [here](https://github.com/acconeer/acconeer-python-exploration/blob/master/src/acconeer/exptool/data/regmap.yaml).

It might be helpful to run Exploration scripts with the -vv (debug log level) flag, for example "python3 examples/utils/ping.py -u /dev/ttyUSB0 -vv". Then you will see all UART traffic logged except for the status polling.

And here's another reply:

[...] PET runs the MS (module server) in streaming mode with UART. SPI uses polling mode, which works fine, but I haven't tried doing the same with UART myself. It should work in the same manner. The steps are

- Configure the service or detector
- ...
- Send main_ctrl create_and_activate
- Poll status until created and activated bits are set
- loop:
  - Poll status until data_ready bit is set
  - Read registers ([...])
  - Send main_ctrl clear_status

Note that some traffic is not logged, for example:
https://github.com/acconeer/acconeer-python-exploration/blob/4ad3eaad69ec4ad808b2acad7bd1e5be7eb57e55/src/acconeer/exptool/clients/reg/client.py#L583

For debugging, I think a good start is to hook up the debug serial port at the top of the XB112. It runs at 3 Mbaud. That's connector J6 in the user guide.