acconeer/acconeer-python-exploration #31: Error with the GUI

lucassbroto, 2020-01-26:
Good evening,

Lately I've been having a big problem with the GUI.

I updated it and now whenever I try to run it, I get the following message in my prompt:
C:\Users\lucas\OneDrive\Área de Trabalho\Tese Radar\acconeer-python-exploration-master >python gui/main.py

Lib version mismatch. Installed: 3.1.1 Latest: 3.2.8
You probably need to reinstall the library (python -m pip install -U --user .)
The GUI might not work properly!

However, when I run this command, I get this:

C:\Users\lucas\OneDrive\Área de Trabalho\Tese Radar\acconeer-python-exploration-master >python -m pip install -U --user
ERROR: You must give at least one requirement to install (see "pip help install")

The folder doesn't have the file required to install. 

But I can still Launch the GUI, but everything runs extremely slow. I am not able to save or to Load any files because the window keeps freezing, while the prompt shows:

SHGetFileInfo() timed out for  C:\Users\lucas\OneDrive\┴rea de Trabalho\Tese Radar\acconeer-python-exploration-master 
SHGetFileInfo() timed out for  C:\Users\lucas\OneDrive\┴rea de Trabalho\Tese Radar\acconeer-python-exploration-master \.github
SHGetFileInfo() timed out for  C:\Users\lucas\OneDrive\┴rea de Trabalho\Tese Radar\acconeer-python-exploration-master \docs
SHGetFileInfo() timed out for  C:\Users\lucas\OneDrive\┴rea de Trabalho\Tese Radar\acconeer-python-exploration-master \examples
SHGetFileInfo() timed out for  C:\Users\lucas\OneDrive\┴rea de Trabalho\Tese Radar\acconeer-python-exploration-master \gui

after that I have to restart the whole thing.

Any idea?

thank you

dj-acconeer, 2020-01-27:
From your logs, it looks like you missed "."  at the end.
Please make sure you type in the full command
`python -m pip install -U --user .`
This will make sure python updates the libraries the GUI is using to communicate with the sensor.

dj-acconeer, 2020-01-28:
Feel free to reopen if this didn't fix the issue for you

closed 2020-01-28 by dj-acconeer