acconeer/acconeer-python-exploration #34: Save to file does not save the latest state of the processing config settings

JNevrly, 2020-02-07:
**Describe the bug**
The "Save to file" function does not save the last state of processing config, it saves the settings state at the moment when the measurement was started. 

**To reproduce**
1. Start measurement in any scan control (e.g. Envelope)
2. In processing settings, change any parameter (eg. History length)
3. Stop the measurement.
4. Click Save to file, save to npz or hd5
5. Reviev the ``processing_config_dump`` section of the saved file, the values will not correspond to the settings in the GUI.

**Environment (please complete the following information):**
 - HW and Host OS independent behavior.
 - v3.2.8

**Additional context**
"Save to file" saves last x measurement frames, but only the "first" processing config state. This is very contradictory/confusing behavior.

dj-acconeer, 2020-02-07:
It should keep the processing config settings from last time you clicked play/replay.
I did a quick check and noticed that it indeed didn't keep the change.
Thanks for pointing this out, we'll look into it.

JNevrly, 2020-02-08:
I did not actually test the replay button. Anyways, my natural expectation (and of course this is subjective) is that when I click on "Save" in any application, ideally the values which I see on the UI are being saved, not values that has been some time ago and I might have changed them since...

After I looked at the code and understanding the inner workings of the Exploration tool better, the way the save is implemented now makes kind of sense. The alternative - making config dumps at the save time, could possibly invalidate the sensor data that are in the recording buffer (lets say user changes the sweep range after clicking stop but before clicking save). So that would be confusing as well ;)

The only "clean" way to do is I think is to separate the treatment of sensor config (which has direct relation to the data in the buffer) and the processing config, which is stored just for reference. E.g. the save button would be disabled once the user changes sensor config after recording stop, but it would not be disabled if the user changes processing settings? Not an easy thing to decide...

dj-acconeer, 2020-03-04:
Changed behavior in commit
https://github.com/acconeer/acconeer-python-exploration/commit/24291c8531058b3edcc30ae90609a43b20d6ae88

When saving data, the current settings from the GUI are used.

closed 2020-03-04 by dj-acconeer