acconeer/acconeer-python-exploration #36: Measuring above 7 meters xm112/xb112

itsonlycode, 2020-03-12:
Trying to evaluate if xm112 can be used to measure distances above the specified max range.
All I need is to be able to measure the distance and no motion detection needed.
I would like to go as far as 15 meters. Is that even possible with xm112 ?

erikmansson, 2020-03-12:
This is technically possible with the A1 radar and, by extension, the XM112. However, this is currently not supported in RSS (the sensor software stack). It is likely that this will be supported in the future, but at this point, I can't give you a time plan for the release.

itsonlycode, 2020-03-13:
Do I understand it correctly that I can measure up to 10m or only to 7 as that seems like the max I can set it to in the examples?

erikmansson, 2020-03-13:
7 m is the current maximum

closed 2020-03-13 by itsonlycode