acconeer/acconeer-python-exploration #39: Error loading training data with more than a few frames

michaelasp, 2020-04-21:
Hi, I'm getting an error loading training data into the ML gui. When I have more than a couple frames in one npy training data file it gives me an error: 

`Failed to load training data: Axis 13 is out of bounds for array of dimension 4.
`

If I put in a file with 3 frames or less it works fine but once I get more than that it just throws that error. the axis number varies but it's always the same error. 

I'm running an A111 if that helps. Thanks!

dj-acconeer, 2020-04-22:
Could you give me some more details to find out what the problem is?
If I understand correctly, you have a few saved ml_session files, which you want to use for training and when you load it, it prints this error?
When did you last pull our repository? We updated some code 2 days ago, which might have introduced some bugs.

michaelasp, 2020-04-22:
I figured out the error after looking through the code, it seems like one of the function calls (numpy.expand_dims) has some undefined behavior that was turned into an error on numpy 1.18. My solution was just to use 1.17 and it works fine now.

dj-acconeer, 2020-04-24:
Thanks for the input. I actually ran across this issue before, but forgot about.
I'll push fix later today.

closed 2020-04-24 by dj-acconeer