acconeer/acconeer-python-exploration #40: GUI doesn't connect with COM port

MuthukumarEC, 2020-04-28:
Hello,
I try to connect my MX122 module in GUI script it provide some error as below i mentioned

Traceback (most recent call last):
  File "gui/main.py", line 1484, in connect_to_server
    info = self.client.connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\base.py", line 29, in connect
    info = self._connect()
  File "C:\Users\PC\AppData\Roaming\Python\Python37\site-packages\acconeer\exptool\clients\reg\client.py", line 199, in _connect
    raise ClientError("could not connect, no response")
acconeer.exptool.clients.base.ClientError: could not connect, no response

erikmansson, 2020-04-28:
I most cases when unable to connect in this way, the firmware isn't flashed properly. Please try flashing again and get back to us.

MuthukumarEC, 2020-04-28:
i fully flashed the module and update DFU in v2.2.1 and run the module. It show same com port error

pthorell, 2020-04-30:
Did you flash the module using nrfutil accoring to the instructions on page 8 in the getting started guide (https://developer.acconeer.com/download/getting-started-guide-xm122-pdf/) ?

MuthukumarEC, 2020-05-04:
I try to flash module via command i had error as bellow

Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 1464, in <module>
  File "click\core.py", line 764, in __call__
  File "click\core.py", line 717, in main
  File "click\core.py", line 1137, in invoke
  File "click\core.py", line 1137, in invoke
  File "click\core.py", line 956, in invoke
  File "click\core.py", line 555, in invoke
  File "nordicsemi\__main__.py", line 1042, in serial
  File "nordicsemi\__main__.py", line 956, in do_serial
  File "nordicsemi\dfu\dfu.py", line 129, in dfu_send_images
  File "nordicsemi\dfu\dfu.py", line 90, in _dfu_send_image
  File "nordicsemi\dfu\dfu_transport_serial.py", line 214, in open
pc_ble_driver_py.exceptions.NordicSemiException: No ping response after opening COM port
[4228] Failed to execute script __main__

pthorell, 2020-05-04:
Make sure that you have put the module in DFU mode (see page 7 in the getting started guide) and check that you are using the right COM port.

erikmansson, 2020-05-13:
Closing this due to inactivity. Please feel free to reopen again if needed.

closed 2020-05-13 by erikmansson