acconeer/acconeer-python-exploration #41: Acconeer_utils

danielle123abc, 2020-05-26:
Hello!

I'm working for my research with https://github.com/Kandidatarbete-Chalmers-MCCX02-19-06/RaspberryPiRadarProgram/tree/master/acconeer-python-exploration-master.
I have also downloaded the EVK file from Acconeer, but when I try to install the supplied Acconeer utilities module from this project (this is an older verion) it doensn't install the client folder. Is there another way to install the folder without using the setup.py?

Best regards,
Danielle

erikmansson, 2020-05-27:
The old recommended way to install acconeer_utils was using
```
python3 setup.py install --user
```
instead of
```
python3 -m pip install --user .
```

Have you tried that?

danielle123abc, 2020-05-27:
Yes I tried that but then he did not install the clients folder. Therefore I put the entire acconeer_utils folder in C: \ Users \ danielle \ AppData \ Roaming \ Python \ Python37 \ site-packages so that it does have the clients folder (I think). Only when I try to run the basic.py I get the error:

`
API version <3 are no longer supperted
Got command:
{
"cmd": "" get_version ",
"api_version": 2
}
`

I suspect that's because the rpi_xc112 of the https://developer.acconeer.com/ is version 2.2.1 and the Exploration Tool Accept SDK and Module SW version 1.5.2?

erikmansson, 2020-05-27:
Yes exactly, you need Exploration Tool v3.x.x to use RSS v2.2.1. I'd recommend updating to the latest version.

danielle123abc, 2020-05-27:
Yes, but I would like to run the old version as the ResPiRadar app from Kandidatarbete-Chalmers-MCCX02-19-06 works on it.
If I run the new version here I get this error:

C: \ Users \ danielle \ acconate \ acconate-python-exploration-master \ RaspberryPiRadarProgram-master \ acconate-python-exploration-master \ examples> python basic.py -s 192.168.178.22
Traceback (most recent call last):
  File "basic.py", line 91, in <module>
    main ()
  File "basic.py", line 46, in main
    client.connect ()
  File "C: \ Users \ danielle \ AppData \ Roaming \ Python \ Python37 \ site-packages \ acconeer_utils \ clients \ base.py", line 17, in connect
    self._connect ()
  File "C: \ Users \ danielle \ AppData \ Roaming \ Python \ Python37 \ site-packages \ acconeer_utils \ clients \ json \ client.py", line 40, in _connect
    raise ClientError ("server error while connecting")
acconeer_utils.clients.base.ClientError: server error while connecting

This is the examples / basic.py of the exploration added to the ResPiRadar -> https://github.com/Kandidatarbete-Chalmers-MCCX02-19-06/RaspberryPiRadarProgram/tree/master/acconeer-python-exploration-master.

erikmansson, 2020-05-27:
Alright and you don't have the older SDK, say v1.10.0?

danielle123abc, 2020-05-27:
No I don't have it

danielle123abc, 2020-05-27:
Thanks! I sent you an email

erikmansson, 2020-05-27:
Closing this but feel free to reopen this issue or create a new one

closed 2020-05-27 by erikmansson