acconeer/acconeer-python-exploration #43: Error running service outside of GUI

roycechung, 2020-06-17:
**Describe the bug**
When I try a run a example service outside of the GUI such as /example/sparse.py I run into the following issue: 

The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

Seems to be an issue with MacOS and multiprocessing, but not sure how to fix it. Everything runs fine in the GUI. I've attached the crash report generated. 
[crash_report.pdf](https://github.com/acconeer/acconeer-python-exploration/files/4795469/crash_report.pdf)

**Environment:**
 - Host OS: [MacOS Catalina 10.15.5]
 - Acconeer hardware: [XM112, Raspberry Pi XC112 + XR112]
 - Acconeer software version: [2.2.1]
 - Exploration version: [v3.6.2]
 - Python: 3.7.7

Thanks,
Royce

erikmansson, 2020-06-18:
I don't have access to a Mac, so could you try a possible fix?

In e.g. `examples/services/sparse.py`, change from
```
if __name__ == "__main__":
    main()
```
to
```
if __name__ == "__main__":
    import multiprocessing
    multiprocessing.set_start_method("spawn")
    main()
```
and run in the same way as before.

roycechung, 2020-06-18:
That fixed the issue.

Thanks,
Royce

erikmansson, 2020-07-20:
Great! We'll consider adding this fix to the library in the future.

closed 2020-07-20 by erikmansson