acconeer/acconeer-python-exploration #44: Error running service

gSajad, 2020-07-01:
**Describe the bug**
When I try to run an example such as \examples\services\envelope.py,  I get the following error messages:

**Traceback**
`PS C:\Users\acconeer\acconeer-python-exploration-master\examples\services> python3 .\envelope.py -spi
10:37:50.578 | WARNING | MainProcess      | acconeer.exptool.clients.base        | old server version - please upgrade server
Traceback (most recent call last):
  File ".\envelope.py", line 79, in <module>
    main()
  File ".\envelope.py", line 26, in main
    session_info = client.setup_session(sensor_config)
  File "C:\Users\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\acconeer\exptool\clients\base.py", line 64, in setup_session
    session_info = self._setup_session(config)
  File "C:\Users\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\acconeer\exptool\clients\reg\client.py", line 514, in _setup_session
    ret = super()._setup_session(config)
  File "C:\Users\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\acconeer\exptool\clients\reg\client.py", line 76, in _setup_session
    self._wait_status(regmap.STATUS_FLAGS.CREATED)
  File "C:\Users\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\acconeer\exptool\clients\reg\client.py", line 126, in _wait_status
    raise ClientError("timeout while waiting for status")
acconeer.exptool.clients.base.ClientError: timeout while waiting for status`

**Environment:**
 - Host OS: [Windows 10]
 - Acconeer hardware: [XM112_PB3v1.1 + XB112PB4v1.2]
 - Acconeer python exploration: [llast version]

dj-acconeer, 2020-07-01:
Could you update your module software and try again?

https://acconeer-python-exploration.readthedocs.io/en/latest/evk_setup/xm122.html#setup-xm122

erikmansson, 2020-07-20:
Closing this due to inactivity. Please feel free to reopen if you still experience the same issue.

closed 2020-07-20 by erikmansson