acconeer/acconeer-python-exploration #45: Error in training model

GideonJagen, 2020-07-02:
Hi!

I'm trying to use your machine learning gui and everything works fine until i press the train button. I then get the following error message:

_Failed to train model!
ml_gui_elements.py
4094
class_weight must contain all classes in the data. The classes {1.0} exist in the data but not in class_weight_

The data i'm trying to train with is range segment features collected with in continuous collection mode.

Best regards
Gideon

dj-acconeer, 2020-07-02:
This happens when the list of labels is not correct (either due to a software bug or due to wrong configuration).

Could you send a screenshot of the message window that pop ups when you load the train files?

GideonJagen, 2020-07-03:
[image]

dj-acconeer, 2020-07-03:
You need at least 2 labels!
When you load the training data, make sure you select all of you training files, containing at least 2 different labels.

GideonJagen, 2020-07-03:
Well that makes sense!

Thank you for your help!

closed 2020-07-03 by GideonJagen