acconeer/acconeer-python-exploration #48: How to properly upgrade the RSS server?

wanglouis49, 2020-07-28:
I am trying to upgrade my RSS server on XM112 from 2.2.1 to 2.4.0 as recommended by the newest version of this repo. I tried to re-flash with BOSSA but it says "could not connect to device on COM3", whereas it can be connected from Acconeer's code on the same port. I wonder if re-flashing is the right way to upgrade? And if yes how can I connect it to BOSSA?

erikmansson, 2020-07-29:
Yes, this is the right way to upgrade. In most cases, either the serial port is hogged by some other app, like Exploration tool, or the device isn't in the bootloader. Try:

- Exit all apps possibly accessing the port, except for BOSSA
- Replug the device
- Hold down the ERASE button
- Push the NRST button
- Release the NRST button
- Let go of the ERASE button
- Try flashing

wanglouis49, 2020-07-29:
Right, the device wasn't in the bootloader. The RSS has been successfully upgraded. Thanks!

closed 2020-07-29 by wanglouis49