acconeer/acconeer-python-exploration #49: Will Ranged Doppler Image be supported?

wanglouis49, 2020-07-28:
As Ranged Doppler Images collected from SOLI have been shown very effective as the input features to the hand gesture recognition network ([paper](https://cse.buffalo.edu/faculty/dimitrio/courses/cse709_s17/material/papers/deep_soli.pdf)), I wonder is there any plan to support this feature or is there any feasible way to derive it from currently available services.

erikmansson, 2020-07-29:
You can derive it from the IQ and sparse services. In both cases, you take the (depthwise) FFT over a number of sweeps. For IQ, this can be any number of sweeps with any overlap. For sparse, we recommend taking the FFT on frames only, like this: 
https://github.com/acconeer/acconeer-python-exploration/blob/ff0c8cf9eef0e10275ec81fb516f7d556d459e1a/examples/processing/sparse_fft.py#L90-L93

 See the [sparse docs](https://acconeer-python-exploration.readthedocs.io/en/latest/services/sparse.html) for more information on sweeps/frames.

[This project from ETH](https://www.researchgate.net/publication/338598690_Low_Power_Embedded_Gesture_Recognition_Using_Novel_Short-Range_Radar_Sensors) is a good example of using the IQ service for gesture recognition.

wanglouis49, 2020-07-29:
Thanks! Will try it out.

closed 2020-07-29 by wanglouis49

erikmansson, 2020-08-10:
In case you're interested, [here's a master's thesis](https://lup.lub.lu.se/student-papers/search/publication/9002607) using range-dopper for hand gesture recognition.