acconeer/acconeer-python-exploration #50: Can't start acc_streaming_server on Raspyberry Pi 4 after kernel update to 5.4.x

aq-daniel, 2020-08-20:
**Short description**
After updating the kernel of my Raspberry Pi 4 to version 5.4.x, the acconeer streaming server won't start and displays a 'could not write to gpio export' error.

**Steps to reproduce**
How to reproduce the behavior:
1. Update linux-kernel from version 4.19.118 to 5.4.x (commands: sudo apt update and sudo apt upgrade)
2. Try to start 'acc_streaming_server_rpi_xc112_r2b_xr_112_r2b_a111_r2c'
3. Streaming App won't start and outputs the error 'Could not write to gpio export: Device or resource busy' (see screenshot below)
4. Can't connect to the sensor via exploration tool or any other application because the streaming server isn't running

**Screenshot**
Error with driver_gpio_linux_sysfs: Could not write to gpio export: Device or resource busy
[image]

**Environment:**
 - Host OS: Win 10 Pro
 - Pi4 OS: Raspian Buster with linux kernel 5.4.59-v7l
 - Acconeer hardware: XC112 + 3* XR112
 - Acconeer software version: 2.4.0
 - Exploration version: 3.7.5

**Additional context and occurence description**
After a standard update routine of Raspian OS with 'sudo apt upgrade' the acc_streaming_server stopped working. The update included the kernel version 5.4.59 and now doesn't allow me to start the acconeer streaming server. If I manually revert the kernel to version 4.9.118 ([this release](https://github.com/Hexxeh/rpi-firmware/tree/e1050e94821a70b2e4c72b318d6c6c968552e9a2)) everything is working fine. 

- Is this a known issue, which the developers of the streaming server are aware of? 
- Is this issue somehow related to the gpio sysfs interface being deprecated for newer kernel releases? And are there any known changes to make in order to get it back up and working? Although I did some search in the raspi-forums ([https://www.raspberrypi.org/forums/](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769)) I couldn't find a gerneral problem with gpios after linux kernel update, i.e. in my opinion it may be a software issue within the streaming server.

I would be thankful for some further investigation. If I missed something or you need any additional information feel free to ask. Thank you!

bankrasrg, 2020-08-20:
Same error here:

- Linux raspberrypi 5.4.51-v7l+
- SparkFun SEN-15577
- Acconeer software version: 2.4.0

None of the examples or reference applications are functional.

johannesacco, 2020-09-01:
Hi,

There are two problems:
1. Our "acc\_board\_\*" files tries to open the [GPIO 8/SPI0 CS0](https://pinout.xyz/pinout/pin24_gpio8) as a GPIO and configure it as an output. In later kernel versions this fails as the kernel spi driver (spi-bcm2835.c) will request ownership of them. This is solved by not trying to configure them in the "acc\_board\_\*" files. (Remove the #define PIN\_SS\_N/PIN_SPI\_SS\_N and the lines using them).
2. For XC112 we need to use the [GPIO 7/SPI0 CS1](https://pinout.xyz/pinout/pin26_gpio7) as a GPIO but the kernel driver will prevent this. A solution to this is to use an overlay to remap it to another GPIO. This can be done by adding "dtoverlay=spi0-cs,cs0_pin=8,cs1_pin=4" to "/boot/config.txt" and reboot.

We will correct this in a coming release and update the documentation. If you like to try it out you can do above changes in an existing delivery.

aq-daniel, 2020-09-03:
@johannesacco Thanks for this quick solution. I tested it and itseems to work for my setup with kernel version 5.4.59. I'll appreciate the integration in an upcoming release and close this issue for now. 

Thank you all for your great work and keep up the regular updates! :)

closed 2020-09-03 by aq-daniel