acconeer/acconeer-python-exploration #54: wrong elfclass

skywo1f, 2020-10-18:
**trying to run the basic example:**
pi@raspberrypi:~/acconeer-python-exploration/examples $ python3 basic.py -spi
Exception raised in SPI communication process:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 623, in run
    self._run()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 641, in _run
    ret = getattr(self, cmd)(*cmd_args)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 694, in connect
    self.dev = libft4222.Device()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 241, in __init__
    _load_dll()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 193, in _load_dll
    dll = ctypes.CDLL(lib_path)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/data/libft4222/libft4222.so.1.4.2.184: wrong ELF class: ELFCLASS64

**but if I ask it 32 or 64 bit:**
pi@raspberrypi:~/acconeer-python-exploration/examples $ uname -a
Linux raspberrypi 5.4.71-v8+ #1352 SMP PREEMPT Fri Oct 16 18:12:52 BST 2020 aarch64 GNU/Linux

erikmansson, 2020-10-18:
For SPI, we only support x86-64, so it won't work on the Pi. However, LibFT4222 seems to support ARM as well, so we'll look into adding it to the tool. But for now, if possible, use UART instead.

closed 2020-11-05 by erikmansson

andrehedesand, 2021-01-13:
Support for most Raspberry Pi devices has been added with the commit `9555f06`.

This should now be resolved.

skywo1f, 2021-04-30:
thanks, that issue went away. I am now seeing a new bug which I will start a new issue for