acconeer/acconeer-python-exploration #56: XE132/XM132 not listed in the usb devices

alexsaso, 2020-10-28:
Dear community,

I have the XE132/XM132 entry module. I have followed the instructions to first flash the module. When I connect the device to the laptop using the micro USB cable (the power switch is switched to USB), the red LED light on the XE132 turns on. However, the laptop does not seem to recognize a new device (see below for the commands I have tried and the results I get). I have tried different micro USB cables and different laptops with Linux Ubuntu 18.04 on it. I always get the same result. The LED light turns on, but the system does not see the new USB device connected.

Did anyone experience similar issue, or does anyone have an idea what might be wrong?

Thankfully,
Alex

**Environment:**
 - Host OS: Ubuntu 18.04
 - Acconeer hardware: XM132 evaluation kit (XE132/XM132)
 - Connection cable: micro USB / USB cable

**Trying to find the serial port:**
I have tried the following commands in linux terminal do find the serial port. None of this commands will list the serial port under which the acconeer device is mounted. 

Note: before using the **dmesg** command, i have cleared its log using the ```sudo dmesg -C``` command. Then I have plugged in the XE132 module and used the ```dmesg``` command, and its output was empty. If I use another device (Lord Microstrain IMU unit) I can immediately see it under the ttyACM0.
 
```
rick@morty:~/working/acconeer$ dmesg | grep USB
```
```
rick@morty:~/working/acconeer$ dmesg | grep tty
```

```
rick@morty:~/working/acconeer$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0461:4e35 Primax Electronics, Ltd 
Bus 001 Device 003: ID 046a:0180 Cherry GmbH 
Bus 001 Device 002: ID 1395:0025 Sennheiser Communications 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
rick@morty:~/working/acconeer$ python -m serial.tools.list_ports
/dev/ttyS0          
/dev/ttyS4          
2 ports found
```

vackraetraed, 2020-10-29:
Hello Alex,

Anton from Acconeer here. 
This issue looks strange, normally 2 ports should show up at /dev/ttyUSB0 and /dev/ttyUSB1

Just to clarify, you have NEVER successfully flashed the module? I.e. this is your first time trying to flash and you can not find the ports.

Best regards,
Anton

alexsaso, 2020-10-29:
Hi Anton,

Thanks for your answer. No, never, I have just received the device on Monday. I have followed the instructions I found on github, readdocs and yt video manual on XM132. The first time I plugged the device into the USB the LED light went on, but I could not find the port for the device (not in Windows not in Linux). The computer does not recognize that any device has been plugged in.

Best regards,
Aleks

vackraetraed, 2020-10-29:
Hi again Alex,

Can you make a ticket at: https://acconeer.freshdesk.com/support/home
Then I will forward this issue to the correct person regarding potentially replacing the device.

Best regards,
Anton

closed 2020-10-30 by alexsaso