acconeer/acconeer-python-exploration #57: calculating RCS from IQ Data

MDkontroller, 2020-11-04:
Hi Acconeer Guys!
Thanks for making this amazing product I have spend a nice time with my Sparkfun :)
I was wondering if could you give as some tips in order to get RCS from A111 radar?
I have checked radar equation and unfortunately I couldnt go further from the avaliable Radar specs
Any advice in that direction?
Thanks in advice ;)

RilleN, 2020-11-05:
Hi MDkontroller,

To get the best measurement of the RCS of an object I would compare its peak height squared in the envelope data with the same figure for an object with known RCS, at the same distance. The most practical object with known RCS is probably a metal sphere and one example of low-cost metal spheres can usually be found as Christmas tree ornaments.

It is preferable to turn off noise normalization during these measurements, since the measurements should be done with the same sensor, to avoid effects from sensor variations.

The magnitude of the IQ data, instead of the Envelope data, should work as well, but with the IQ service, the sensor and processing are optimized for accuracy in the phase, not the amplitude.

MDkontroller, 2020-11-06:
Hi erikmansson!
Thanks for your advice! I think its a good experiment.  I will first measure any fixed RCS around so can check this forward!

erikmansson, 2020-11-06:
Thanks @RilleN :) Closing this for now, feel free to reopen if you have further questions.

closed 2020-11-06 by erikmansson