acconeer/acconeer-python-exploration #58: No executable file after *make

MDkontroller, 2020-11-16:
Hi guys

Lately I have been doing some scrips for my Sparkfun, after including all headers and run make no executable is in the out file. However I have change the file name to "example_service_iq.c" and then a executable is generated and works fine.
It is possible to generate my own named executable files? 
All best ;)

vackraetraed, 2020-11-16:
Hello there,
Anton from Acconeer here.

It is possible for you to create your own executable. I am guessing you are using a raspberry pi?
Our makefile for the raspberry uses rules that can be found in the "rule" folder. As an example you can see: "rule/makefile_build_example_service_iq_a1r2_rpi_xc112_r2b_xr112_r2b_a111_r2c.inc"
You need to add your own rule for your program and alter the file.
I would copy the previous files to keep the build system intact and simply add my new program to the build.

Best regards,
Anton

MDkontroller, 2020-11-19:
Ok, many thanks vackraetraed!
I think I can get it done :)

closed 2020-11-19 by erikmansson