acconeer/acconeer-python-exploration #59: Compute Distance from data

DavidBrandes, 2020-11-18:
Hey,
I would like to understand the the Distance computation from the Client data a bit better. I could not find the place in your code where this is happening.

So what I get from the Client in one data packet are sweeps of different amplitudes. If I now assume that the value with the biggest amplitude corresponds to the wanted distance, can I then compute it as:

distance = range_start_m + step_length_m * argmax(data) ?

This method seems to give correct measurements, however I would like to assure that this is in fact the proper way to compute the distance.

erikmansson, 2020-11-18:
Yes, that is a proper way to compute the distance. If you need a method of finding any number of peaks, including zero, check out the distance detector. It's in the GUI and you can find the code here:
https://github.com/acconeer/acconeer-python-exploration/blob/master/examples/processing/distance_detector.py

DavidBrandes, 2020-11-18:
Thank you for the quick reply!

I noticed that when I change the range for the radar (e.g. from [0.2m,1.2m] to [0.5m,1m]) the distance measurement starts to change slightly by one millimeter or two. Does this imply that for smaller ranges the measurement is getting more precise, i.e. the radar can differentiate between more distance steps? Or is there no such implication?

erikmansson, 2020-11-19:
No, the resolution stays the same regardless of range. It is set by the downsampling factor.

Is the change in distance consistent going between the two configurations?

DavidBrandes, 2020-11-19:
It was consistent. However I checked it again and now the measured difference between both settings is only marginal. So I will close this issue.

Thank you for your help, I appreciate it!

closed 2020-11-19 by DavidBrandes