acconeer/acconeer-python-exploration #60: Sampling Frequency ()

MDkontroller, 2020-11-19:
Hi guys!
this not an issue, just a question from a Student :)
I have read in Acconeer docs, sampling frequency of the radar is not given just a wavelenght
which assuming the formula:( f = c / lambda) are we talking about 120GHz sampling rate?
Im very impresive considering another radars like FMCW where tipically is about Mhz range.
Thank u in advance!

erikmansson, 2020-11-19:
The highest effective resolution the A1 can achieve is ~3 ps, corresponding to ~0.5 mm or ~300 GHz. While certainly an impressive number, note that this is the sample resolution of the reconstructed signal and not the actual sample rate. The sample rate, usually referred to as pulse repetition frequency or PRF for short, is 13 MHz.

For some more information, see the [radar sensor introduction](https://acconeer-python-exploration.readthedocs.io/en/latest/sensor_introduction.html).

MDkontroller, 2020-11-20:
Hi erik!
thanks for your information! however I’m just bit confuse. I assume radar transmitter send this 1000 pulses at a frequency of 60 Ghz and then ADC converter sample the signals at 13MHz?
im my current experimet and saving some seconds of IQ data recording with the raspi API in csv, and wanted to make a some nice plots of PSD , FFT and so on in python. therefore the sampling frequency :)
we want to implement sparkfun in our University lab to introduce students to radar tecnology

erikmansson, 2020-11-23:
Yes. In every 13 MHz period, a 60 GHz pulse is sent, received, and sampled. For every data point corresponding to one depth, the radar sends one or several pulses depending on HWAAS, profile, service, and sampling mode.

MDkontroller, 2020-11-25:
HI erik!
Thank for your help understanding this radar,
as being fs = 13MHz the sampling frequency, its defined from that a frequency resolution of fres = fs / 847 = 15.3KHz, being 847 the number of elements per sweep in iq data measurement.
I was wondering in this point if, its possible to increase the size of the sweep?
the idea in background is measure frequencies from objects in motion from iq data by performing an fft of iq amplitude.

erikmansson, 2020-11-26:
Almost. Every point of the 847 takes a number of pulses to measure so your fres is substantially lower in reality.

Increasing the size of the sweep is just a matter of extending the range measured. The resolution is ~0.5 mm per point.

While that method of measuring velocity is _possible_,  we typically do so by taking the FFT over a number of sweeps instead of over the depth/sample dimension. See https://github.com/acconeer/acconeer-python-exploration/blob/master/examples/processing/obstacle_detection.py for an example of that using the IQ service.

MDkontroller, 2020-11-30:
Hi erik!
many thanks for your explanation again,
we will make some measurements and check what we see from them
for us is important to know what happens "in background" to be able to explain it to our colleges studying signal processing, many thats in this way :)

erikmansson, 2020-12-02:
No problem! Closing this for now. Please feel free to reopen or create new issues if you have further questions.

closed 2020-12-02 by erikmansson