acconeer/acconeer-python-exploration #64: Failed to train model

whizz5, 2021-01-01:
Unable to train model using default layers

I have loaded training data for a model and after configuring the features to inspect and the model parameters (using default layers from  "default_layers_2d.yaml" I press the "train" button. I am then presented with an alert (see attached picture) stating that training failed but I can't figure out how to resolve it. I'm aware that when using the default layers, I have to remove Conv2D and MaxPool2D as my feature frames are smaller than the size outlined in the docs. 

I will also provide the training data I used which should help trying to reproduce this issue (in the zip file)

 - Host OS: Windows
 - Acconeer software version: 2.7.1
 
Not too sure if it's a set up issue or a program bug, thanks in advance

[image]

[ml_session_data_2021_01_01_1728_swipe_left.zip](https://github.com/acconeer/acconeer-python-exploration/files/5759277/ml_session_data_2021_01_01_1728_swipe_left.zip)

dj-acconeer, 2021-01-01:
Hi  whizz5,
what version of Tensorflow and Keras are you using?

whizz5, 2021-01-01:
Update, added another data set and I was able to proceed with the training, another ticket mentioned that 2 data sets are needed

whizz5, 2021-01-01:
> Hi whizz5,
> what version of Tensorflow and Keras are you using?

Keras version 2.4.3
Tensorflow version 2.4

closed 2021-01-01 by whizz5

dj-acconeer, 2021-01-01:
Tensorflow 2.x support is still experimental, but I'll try to help you sort this out.

One more thing, how many labels do you use? You need at least 2 different labels.

whizz5, 2021-01-01:
Initially, I only had one label however I added another one which resolved my issue

whizz5, 2021-01-01:
@dj-acconeer Thank you for all the work you and your colleagues have done getting this tool up and running