acconeer/acconeer-python-exploration #65: Bad Config error

MuthukumarEC, 2021-01-13:
I have XM122 Evm I tested to GUI of Acconeer explorer tool. while I'm using Distance detector or any of these testing function choose GUI to show bad config like the below image
using python version is python 3.7.3

[image]

erikmansson, 2021-01-13:
Can you run anything at all, for example the sparse example with default config?

MuthukumarEC, 2021-01-13:
Thanks for your reply. I have check sparse examples it's working fine. but some other example doesn't work properly. Below mentioned examples are don't work its show the same error in "Bad Config".

envelope
IQ
Power bins
phase tracking
sleep breathing
Obstade detedction(IQ)
Button Press(envelope)
Distance Detector
Parking(envelope)

erikmansson, 2021-01-13:
Ok, thanks. What SDK and Exploration tool versions?

MuthukumarEC, 2021-01-13:
Below I mentioned the version of SDK and Explorer
Acconeer XM122 EVM SDK Version is 2.7.1
Acconeer Exploration Tool Version is 2.7.1
python version is 3.7.3
windows 10 64 bit os

vackraetraed, 2021-01-13:
Hello MuthukumarEC,

Do you have the possibility to flash the envelope example "example_service_envelope" from our SDK and captures the logs from the USB at 115200 baudrate? Then paste the logs here.

MuthukumarEC, 2021-01-13:
How to flash the envelope example in my XM122 EVM. I don't know how to flash envelope.py code in my XM122 is there are any steps available please send me.

vackraetraed, 2021-01-13:
Hello MuthukumarEC,

That is the python exploration tool examples used together with the module software flashed on the module.

I would like you to download the XM122 SDK from our https://developer.acconeer.com and setup the development environment: https://developer.acconeer.com/download/xm122-software-development-guide-pdf/

There should be precompiled firmware for the envelope example which you can flash using nrfutil as explained in the guide.
Then you can capture the logs on the USB at 115200 baudrate.

Best regards,
Anton

MuthukumarEC, 2021-01-13:
Hello @vackraetraed,

Thanks for your support as per following the software development guide i have upload "example_service_envelope.Zip" file from "acconeer_xm122" through DFU in the nrf_tool_box app. Finally, the device sends some error messages by UART as below

<info> app: Starting example application

<info> app: Acconeer software version v2.7.1

<info> app: rss:Radar system software activated

<info> app: base_configuration:sensor 1 config: 10 1 11 6 7 9 READY A 0 0 0

<info> app: cpd_cbank_and_vana_calibration:Result: (4, 0)

<error> app: sensor_manager:Calibration failure for sensor 1

<info> app: acc_service_create() failed

<info> app: rss:Radar system software deactivated

vackraetraed, 2021-01-13:
Hello @MuthukumarEC ,

Have the envelope or IQ services worked for you before or have they always failed when creating the service?

Best regards,
Anton

MuthukumarEC, 2021-01-13:
Hello @vackraetraed,

Always I got this same problem every time. I test this module for development purposes but this device doesn't work in GUI.

vackraetraed, 2021-01-13:
It sounds like the sensor on the XM122 you are using is broken.

Do you have any other XM122 available which you can try?

Best regards,
Anton

erikmansson, 2021-02-10:
Closing this due to inactivity. Please feel free to reopen.

closed 2021-02-10 by erikmansson