acconeer/acconeer-python-exploration #66: Multiple Sparkfun Sensors

SebiLu, 2021-01-23:
Hi, i already set up a working system, measuring with the envelope Service. 
I am using the Nucleo-64 with the STM32l476 and a single Sparkfun A111 board. I got everything running smoothly.

Recently I wanted to upgrade my system with a second Sensor, changed the hal integration file for the "...stm32cube_2x_sparkfun_a111.c" and changed all the neccessary pin names. 
I can build the project without errors or warning, so I guess the basics are setup correctly. 
But when I run the microcontroller in debug mode and watch the output in the console, it displays the following: 

0:00:00.000 (I) (rss): Radar system software activated
00:00:00.001 (I) (base_configuration): sensor 1 config: hu 10 1 17 15 0  E READY 134295232 0 0
00:00:00.005 (E) (sensor_protocol): Expected asic_id = 0x1112 [Read asic_id = 0x0]
00:00:00.005 (E) (sensor_manager): Sensor 1 does not communicate properly, is it connected?
00:00:00.009 (I) (base_configuration): sensor 2 config: hu 10 1 17 15 0  E READY 134295232 0 0
00:00:00.013 (E) (sensor_protocol): Expected asic_id = 0x1112 [Read asic_id = 0x0]
00:00:00.013 (E) (sensor_manager): Sensor 2 does not communicate properly, is it connected?
acc_service_create() failed
00:00:00.017 (I) (rss): Radar system software deactivated

Since I cannot find any documentation to a multiple Sensor setup, other than the fact, that a maximum of two sensors is allowed per SPI-Bus (which I chose to do) I hope to find some help over here.

Thanks in advance!

OlaAcconeer, 2021-01-24:
Hi SebiLu,

May I ask you if you made any pin changes for the first sensor when you decided to add support for the second sensor.

> Since I cannot find any documentation to a multiple Sensor setup, other than the fact, that a maximum of two sensors is allowed per SPI-Bus (which I chose to do) I hope to find some help over here.
As far as I know, we don't have any restrictions regarding this. On a certain SPI-bus there might be support only for two hardware chip selects but then you can always trigger the slave select in software (even though it might not be preferable.)

If you want to se a multiple sensor setup, though it's been made for raspberry pi and spidev, check out the XC112 board by loging innto developer.acconeer.com -> Software downloads -> XC112

Don't give up, we will help you to make it work! :)

Btw, do you have access to a logic analyzer?

OlaAcconeer, 2021-01-24:
// This is how we solved two sparkfun sensors on an STM device.

// Copyright (c) Acconeer AB, 2020
// All rights reserved
// This file is subject to the terms and conditions defined in the file
// 'LICENSES/license_acconeer.txt', (BSD 3-Clause License) which is part
// of this source code package.

#include <stdarg.h>
#include <stdbool.h>
#include <stdint.h>
#include <stdio.h>
#include <string.h>

#include "main.h"

#include "acc_definitions.h"
#include "acc_hal_definitions.h"
#include "acc_hal_integration.h"

/* spi handle */
extern SPI_HandleTypeDef A111_SPI_HANDLE;

/**
 * @brief The number of sensors available on the board
 */
#define SENSOR_COUNT 2

/**
 * @brief Max buffer size for the RSS logging
 */
#define LOG_BUFFER_MAX_SIZE 150

/**
 * @brief Size of SPI transfer buffer
 */
#ifndef A111_SPI_MAX_TRANSFER_SIZE
#define A111_SPI_MAX_TRANSFER_SIZE 65535
#endif

/**
 * @brief The reference frequency used by this board
 *
 * This assumes 26 MHz on the Sparkfun A111 Board
 */
#define ACC_BOARD_REF_FREQ 26000000

/**
 * @brief Format for the RSS logging
 *
 */

#define LOG_FORMAT "%02u:%02u:%02u.%03u (%c) (%s): %s\n"

static inline void disable_interrupts(void)
{
	__disable_irq();
}

static inline void enable_interrupts(void)
{
	__enable_irq();
	__ISB();
}

//----------------------------------------
// Implementation of RSS HAL handlers
//----------------------------------------

static void acc_hal_integration_sensor_transfer(acc_sensor_id_t sensor_id, uint8_t *buffer, size_t buffer_size)
[... 161 lines cut ...]
}

static const acc_hal_t hal =
{
	.properties.sensor_count          = SENSOR_COUNT,
	.properties.max_spi_transfer_size = A111_SPI_MAX_TRANSFER_SIZE,

	.sensor_device.power_on                = acc_hal_integration_sensor_power_on,
	.sensor_device.power_off               = acc_hal_integration_sensor_power_off,
	.sensor_device.wait_for_interrupt      = acc_hal_integration_wait_for_sensor_interrupt,
	.sensor_device.transfer                = acc_hal_integration_sensor_transfer,
	.sensor_device.get_reference_frequency = acc_hal_integration_get_reference_frequency,

	.os.mem_alloc = malloc,
	.os.mem_free  = free,
	.os.gettime   = acc_hal_integration_get_current_time,

	.log.log_level = ACC_LOG_LEVEL_INFO,
	.log.log       = acc_hal_integration_log
};

const acc_hal_t *acc_hal_integration_get_implementation(void)
{
	return &hal;
}

OlaAcconeer, 2021-01-24:
`
hej

hej
`

OlaAcconeer, 2021-01-24:
`fkdskfldskölfsd`

OlaAcconeer, 2021-01-24:
`

`

OlaAcconeer, 2021-01-24:
[acc_hal_integration_single_thread_stm32cube_2x_sparkfun_a111.zip](https://github.com/acconeer/acconeer-python-exploration/files/5862487/acc_hal_integration_single_thread_stm32cube_2x_sparkfun_a111.zip)

OlaAcconeer, 2021-01-24:
I just uploaded a file which is an integration made by us to support two sparkfun cards on an STM hardware.

SebiLu, 2021-01-25:
Hi, 
Thank you already for the help!!

To give you a quick overlook over the changes I made: 
- I swaped the "standard" single sparkfun integration file for the dual-sparkfun integration file. 
- I let the pin-configuration for the first sensor as it was, but changed the naming of the pins (the new integration file asked for a different naming)
- I added pins for the second sparkfun board (CS, EN, INT)

That should be it I think...

Thanks for the hint, I will take a look at the XC112 project!

I dont have access to a logic analyzer that could handle the speed of the SPI communication, but my DSO can decode SPI.

Thanks again, and I'm looking forward to your response!
Regards,
Sebastian

SebiLu, 2021-01-25:
I acquired two pictures, of the communication, it seems like the chip just wont answer....

[image]
[image]

SebiLu, 2021-01-25:
The Enable Pin is also pulled High directly after Power-up, and pulled low after the unsuccessful SPI communication is done.
So as far as I can tell, all the necessary prerequisits are setup correctly....

SebiLu, 2021-01-25:
Oh come on, I'm somewhat retarded, I didn't plug in Vdd correctly....
I'm so sorry for wasting your time....Its communicating now as it should...

Although thank you for your help!
Regards
Sebastian

closed 2021-01-25 by SebiLu

OlaAcconeer, 2021-01-26:
No worries, who has'nt done that mistake? And I guess we will all do it again :)

I'm happy it all worked out!