acconeer/acconeer-python-exploration #69: difference between amplitude of iq and the amplitude from the envelope service

kangc0016, 2021-02-03:
This is not a bug but trying to understand different services. 

Question 1: If I use the same configuration (profile, HWAAS averaging, etc) for envelope service and iq service, is the amplitude of the iq data equivalent to the envelope data? If not, why are they different? Based on my testing they look different, I want to understand why they are different. 

Question 2: is HWAAS averaging the iq data or is HWAAS averaging amplitude data?

Question 3: Can I reconfigure the device to different profile at run time?

erikmansson, 2021-02-03:
1: The amplitude of the IQ data is not equivalent to the envelope data. This is because the sensor uses a different method to measure phase coherent data.

2: HWAAS averages on the raw data, which has the same effect as averaging on the IQ data.

3: With Exploration Tool no, but with the SDK it's possible to switch between configurations rather quickly.

kangc0016, 2021-02-04:
Thanks, for envelope service, is HWAAS doing coherent integration or non-coherent integration, or in another word, what processing gain should I expect to get from HWAAS averaging?

Regarding switching between configurations, does it take one update time or does it take a constant time no matter what the update rate is?

erikmansson, 2021-02-05:
The integration is coherent, but for the envelope service the SNR gain is not linear. For example, using profile 2, going from HWAAS 10 to 63 results in a typical SNR gain of ~3.7 dB.

Switching between configurations takes a constant amount of time, mostly dependent on the service and MCU used.

closed 2021-02-10 by erikmansson