acconeer/acconeer-python-exploration #73: Modification of base-Service while using Presence detector

SebiLu, 2021-03-22:
Hi,

i am currently working with the presence detector in C not in python, but since I didn't know where to go with my question, I hope to find help here...
But I know would want to modify certain parameters in the Sparse Service, which the Presence detector bases on.  (maximize signal attenuation,...)

I found a function, to modify the base configuration of a service in acc_service.h, but no such thing for the base of a detector...

Since (for example) the maximize signal attenuation parameter is configurable in the Exploration tool while using the presence detector, I guess it should be configurable for me too...

Thanks in advance,
BR, Sebastian

SebiLu, 2021-03-22:
Edit: I did find the "acc_detector_presence_configuration_set_service_data_callback" but it didn't work for me, so maybe there is an example available or something?

PerAtleviAcconeer, 2021-03-22:
Hi Sebastian,

The generic configuration functions that can be set for all services are located in 'acc_service.h'. Service specific configuration functions can be found in acc_service_sparse.h, acc_service_envelope.h etc.

The detectors do not have any generic configuration functions, they are all located in the detector specific files acc_detector_presence.h, acc_detector_distance.h etc.

The 'maximize_signal_attenuation' is not a configuration that is propagated to the presence detector API.

What do you need the setting for? Perhaps there is another way solve your case?

SebiLu, 2021-03-22:
Ah ok I See...thank you for your quick response!

I would like to detect a "presence" very close to the sensor, to do that I would need to activate the maximize signal attenuation due to the high noise levels...the presence detector in combination with the exploration tool worked best for me / my use-case.

SebiLu, 2021-03-22:
I thought using the presence detector would be the " lazy way" without the need to figure out any of the math myself ;-)
But if the generic functions are not available in the detectors, I guess I need to step back to the sparse service anyways.

PerAtleviAcconeer, 2021-03-22:
The most common configurations are there for the detectors (start, length, gain, power save mode etc.), so in general it should definitely be possible to use the detectors in C instead of implementing them yourself.

Perhaps it would be enough to use the gain setting in the presence detector?

SebiLu, 2021-03-22:
Yeah you're right, that might be enough.

So there is no way to create the sparse config, set everything as I would like it to and then use the sparse config to convert it to the presence detector?
This is kinda frustrating to me, because while debugging my code, I can see the "maximize signal attenuation" parameter in the variable window, but cannot access it...but ok.

PerAtleviAcconeer, 2021-03-22:
Not as of today. The reason is that some configurations for the underlying service may not work or will behave strangely when used in the detector. To avoid that, we have limited the amount of configuration that you can do on a detector to what we think is relevant.

Of course, if we get input regarding useful settings to have in the detectors that aren't there now, we will add them if possible.

SebiLu, 2021-03-22:
Sure, I understand that.

Well, then I would like to propose the integration of the "maximize_signal_attenuation" parameter, if that would somehow be possible for you!

I now tried to reduce the gain setting, even to zero, but I keep getting "Data saturated" outputs in my Output window.

I set a couple other influencing factors according, so I might reduce data saturation: Sampling mode A, HWAAS = 1, Gain = 0
Are there other things I could set to improve my situation?

PerAtleviAcconeer, 2021-03-22:
Have you tried changing the profile? Profile 1 should give you the least amplitude. Otherwise, I don't think there are any other settings. I should note that the presence detector might still work even if you get data saturated (but it's not recommended of course).

I will bring up the possibility to set 'maximize_signal_attenuation' in the presence detector.

SebiLu, 2021-03-22:
You are right, I still had profile 2 chosen...now it actually works without data saturation, I kept the settings I mentioned before!

For now, thanks a lot for your Help!

Regards, 
Sebastian

closed 2021-04-06 by erikmansson