acconeer/acconeer-python-exploration #75: XE132 USB device not showing

luisvillaran, 2021-04-06:
I received a brand new XE132 board a few days ago from Digikey.  I tried following the start guides but can't get passed the first step of getting the board to show up as a USB device.  I get the following in the Device Manager:

[image]

I've tried multiple USB cables (I saw someone had an issue with that) and I tried installing the STM32 VCP drivers to no avail.  Any suggestions?

 - Host OS: Windows 10
 - Acconeer hardware:  XE132

erikmansson, 2021-04-06:
Can you try [these](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) drivers?

luisvillaran, 2021-04-06:
> Can you try [these](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) drivers?

The Universal Windows 10 driver on that page worked for me.  Thanks for your help!

closed 2021-04-06 by luisvillaran

erikmansson, 2021-04-06:
Great!