acconeer/acconeer-python-exploration #77: XM132: Baudrate not reseting after reset

dpeterjohnbanner, 2021-04-16:
I have a custom board layout with a host MCU STM32F446 interfacing with the XM132 module over UART.

The firmware running on the STM32F446 has a configuration sequence at startup to configure the XM132. 

Configuration Sequence:
1. Pull NRESET pin low on XM132 to reset
2. Wait for 100ms (just being extra lenient with timing)
3. Pull NRESET pin high on XM132
4. Wait for 100ms (waiting at least 50ms before communicating with the XM132 as per the manual on the XM132)
5. STM32F446's UART baudrate is set for 115,200. (XM132's default baudrate)
6. ReadRequest for "ProductMaxUartBaudrate" (result is 1,000,000 baud)
6. WriteRequest to set XM132's UART baud rate to 1,000,000.
7. Configure STM32F446's UART baud to 1,000,000.
8. Continue configuring XM132 with other settings... etc.

What I have found is that after a power cycle, when I debug the STM32F446's firmware. I can debug just fine and the startup sequence behaves as expected. But if I restart the debug session, the XM132 will not communicate anymore at the default baud rate, even though it is being reset with the NRESET pin. I added some test code and found the XM132 is still operating at the 1,000,000 baud it was set to in the previous debug session.

Any ideas about this? Is this intended behavior? This could be symptom of something wrong with my board layout for the STM32F446, but I want to verify what intended behavior of NRESET pin.

vackraetraed, 2021-04-19:
Hello @dpeterjohnbanner,

Firmware-wise the baudrate should have been reset to the default value when XM132 is restarted.
How are you pulling NRESET low and then setting it high? Are you using a GPIO from the STM32 board? What logic level does the STM32 board use?

Best regards,
Anton

dpeterjohnbanner, 2021-04-21:
Thanks @vackraetraed for the reply.

It turned out to be a bad solder connection on the NRESET pin. I appreciate your information since it helped narrow down the problem.

closed 2021-04-21 by dpeterjohnbanner