acconeer/acconeer-python-exploration #78: rpi4 pyqt5 getpointer called with incorrect name

skywo1f, 2021-04-30:
pi@raspberrypi:~/acconeer-python-exploration/examples $ python3 basic.py 
Traceback (most recent call last):
  File "basic.py", line 1, in <module>
    import acconeer.exptool as et
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/__init__.py", line 6, in <module>
    from . import clients, configs, recording, utils
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/__init__.py", line 2, in <module>
    from .mock.client import MockClient
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/mock/client.py", line 8, in <module>
    from acconeer.exptool.configs import BaseServiceConfig
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/configs.py", line 4, in <module>
    from acconeer.exptool import utils
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/utils.py", line 19, in <module>
    import pyqtgraph as pg
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/home/pi/.local/lib/python3.7/site-packages/pyqtgraph/Qt.py", line 140, in <module>
    from PyQt5 import QtSvg
ValueError: PyCapsule_GetPointer called with incorrect name

skywo1f, 2021-04-30:
$ pip3 list | grep -i pyqt
PyQt5             5.15.2
PyQt5-sip         12.8.1
pyqtgraph         0.12.1

skywo1f, 2021-04-30:
same with
pip3 list | grep -i pyqt
PyQt5             5.15.4
PyQt5-sip         12.8.1
pyqtgraph         0.12.1

skywo1f, 2021-04-30:
I tried a fresh install but now it does not seem to be able to install pyqt5 greater than version 5.11 on the rp (using apt install python3-pyqt5). Any pip installs fail due to not finding a version for the rpi

erikmansson, 2021-05-04:
Getting the graphical stuff running on a Raspberry Pi could be a bit difficult - we generally don't recommend it. I see you're trying to run the basic example which actually doesn't need pyqtgraph to run, so if you only need that then I'd simply uninstall pyqtgraph to keep it from trying to initialize.

closed 2021-05-16 by erikmansson

skywo1f, 2021-05-29:
In that case I see:

pi@raspberrypi:~/acconeer-python-exploration/examples $ python3 basic.py -spi
Exception raised in SPI communication process:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 628, in run
    self._run()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 646, in _run
    ret = getattr(self, cmd)(*cmd_args)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 700, in connect
    self.dev.open_ex()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 255, in open_ex
    check_status(funs["FT_OpenEx"](_locationId, OPEN_BY_DESCRIPTION, byref(self.handle)))
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 236, in check_status
    raise LibFT4222Error("Error {}: {}".format(err, msg))
acconeer.exptool.libft4222.LibFT4222Error: Error 2: DEVICE_NOT_FOUND

Traceback (most recent call last):
  File "basic.py", line 89, in <module>
    main()
  File "basic.py", line 44, in main
    client.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 506, in _connect
    self.__cmd_proc("connect")
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 609, in __cmd_proc
    raise ClientError("exception raised in SPI communcation process")
acconeer.exptool.clients.base.ClientError: exception raised in SPI communcation process
pi@raspberrypi:~/acconeer-python-exploration/examples $ python3 basic.py -spi
Exception raised in SPI communication process:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 628, in run
    self._run()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 646, in _run
    ret = getattr(self, cmd)(*cmd_args)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 700, in connect
    self.dev.open_ex()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 255, in open_ex
    check_status(funs["FT_OpenEx"](_locationId, OPEN_BY_DESCRIPTION, byref(self.handle)))
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/libft4222.py", line 236, in check_status
    raise LibFT4222Error("Error {}: {}".format(err, msg))
acconeer.exptool.libft4222.LibFT4222Error: Error 2: DEVICE_NOT_FOUND

Traceback (most recent call last):
  File "basic.py", line 89, in <module>
    main()
  File "basic.py", line 44, in main
    client.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 506, in _connect
    self.__cmd_proc("connect")
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/clients/reg/client.py", line 609, in __cmd_proc
    raise ClientError("exception raised in SPI communcation process")
acconeer.exptool.clients.base.ClientError: exception raised in SPI communcation process

I did activate spi on raspi-config. Also I ran the example_assembly_test successfully from the rpi_sparkfun library.

erikmansson, 2021-06-01:
Sorry it's perhaps a bit unclear - SPI is for connecting to the XM112. Exploration Tool cannot directly connect to the A111 on a Raspberry Pi.

skywo1f, 2021-06-02:
ah, I am barking up the wrong tree.