acconeer/acconeer-python-exploration #80: issue connecting to serial port and SPI on MacBook

Ondobaka, 2021-05-06:
Hello all, 
I'm having issues connecting the exploration tool gui as well as the breathing examples called from the terminal to the XB-112 module.

Any help would be highly appreciated to get going with these exciting and well built tools. 

Sasha

**To reproduce**
Steps to reproduce the behavior:
I'm selecting the appropriate interface (/dev/cu.usbserial (XB-112)) and hitting connect. This produces the following error on a 2014 MacBook Pro. I am running the most recent software, downloaded yesterday. 

When using the terminal and specifying the above port, I could get the envelope.py working fine, yet not any of the breathing examples. Also, when trying to specify -SPI in a terminal call it faults with the following message at the end: acconeer.exptool.clients.base.ClientError: exception raised in SPI communcation process.

**Traceback**
Traceback (most recent call last):
  File "/Users/sasha/Downloads/acconeer-python-exploration-master/gui/main.py", line 1535, in connect_to_server
    info = self.client.connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 199, in _connect
    self._link.connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/links.py", line 234, in connect
    raise LinkError("failed to connect, timeout")
acconeer.exptool.clients.links.LinkError: failed to connect, timeout

**Environment (please complete the following information):**
 - Host OS: BigSur 11.1
 - Acconeer hardware: [e.g. XB112]
 - Acconeer software version: [most recent]
 - Exploration version: [most recent]

erikmansson, 2021-05-07:
Hi! To clarify, UART works with envelope.py but not with the breathing examples nor the GUI? What output do you get from running a breathing example? Add the flag -vv to get more output.

I'm unsure if SPI works at all with macOS. If you want, post a full traceback from the exception so we can have a look.

Ondobaka, 2021-05-07:
Thanks Erik. The assessment of the issue is correct. Here's the output from breathing.py:

sasha@Sashas-Mac acconeer-python-exploration-master % python3 examples/processing/breathing.py -u /dev/cu.usbserial-01336 -vv
09:24:50.964 | DEBUG   | MainProcess      | acconeer.exptool.utils               | logging configured
09:24:51.681 | DEBUG   | MainProcess      | acconeer.exptool.clients.links       | connect - flow event was not set (timeout)
09:24:52.682 | INFO    | MainProcess      | acconeer.exptool.clients.links       | forcing disconnect...
09:24:52.690 | INFO    | MainProcess      | acconeer.exptool.clients.links       | forced disconnect successful
Traceback (most recent call last):
  File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 345, in <module>
    main()
  File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 30, in main
    session_info = client.setup_session(sensor_config)
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 59, in setup_session
    self.connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 199, in _connect
    self._link.connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/links.py", line 234, in connect
    raise LinkError("failed to connect, timeout")
acconeer.exptool.clients.links.LinkError: failed to connect, timeout
sasha@Sashas-Mac acconeer-python-exploration-master %

Ondobaka, 2021-05-07:
Regarding SPI, I think you were right that macOS is not supported. But, just for the completeness, here's the output of running breathing.py with the -spi option:

sasha@Sashas-Mac acconeer-python-exploration-master % python3 examples/processing/breathing.py -spi -vv
10:15:24.170 | DEBUG   | MainProcess      | acconeer.exptool.utils               | logging configured
10:15:24.182 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sending cmd to proc: connect
Exception raised in SPI communication process:

Traceback (most recent call last):
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 628, in run
    self._run()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 646, in _run
    ret = getattr(self, cmd)(*cmd_args)
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 699, in connect
    self.dev = libft4222.Device()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/libft4222.py", line 250, in __init__
    _load_dll()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/libft4222.py", line 217, in _load_dll
    raise RuntimeError("OS not supported")
RuntimeError: OS not supported

Traceback (most recent call last):
  File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 345, in <module>
    main()
  File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 30, in main
    session_info = client.setup_session(sensor_config)
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 59, in setup_session
    self.connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 506, in _connect
    self.__cmd_proc("connect")
  File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 609, in __cmd_proc
    raise ClientError("exception raised in SPI communcation process")
acconeer.exptool.clients.base.ClientError: exception raised in SPI communcation process
sasha@Sashas-Mac acconeer-python-exploration-master % 

Many thanks for your help!

erikmansson, 2021-05-10:
> 
> 
> Thanks Erik. The assessment of the issue is correct. Here's the output from breathing.py:
> 
> sasha@Sashas-Mac acconeer-python-exploration-master % python3 examples/processing/breathing.py -u /dev/cu.usbserial-01336 -vv
> 09:24:50.964 | DEBUG | MainProcess | acconeer.exptool.utils | logging configured
> 09:24:51.681 | DEBUG | MainProcess | acconeer.exptool.clients.links | connect - flow event was not set (timeout)
> 09:24:52.682 | INFO | MainProcess | acconeer.exptool.clients.links | forcing disconnect...
> 09:24:52.690 | INFO | MainProcess | acconeer.exptool.clients.links | forced disconnect successful
> Traceback (most recent call last):
> File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 345, in
> main()
> File "/Users/sasha/Downloads/acconeer-python-exploration-master/examples/processing/breathing.py", line 30, in main
> session_info = client.setup_session(sensor_config)
> File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 59, in setup_session
> self.connect()
> File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/base.py", line 30, in connect
> info = self._connect()
> File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/reg/client.py", line 199, in _connect
> self._link.connect()
> File "/Users/sasha/Library/Python/3.9/lib/python/site-packages/acconeer_exptool-3.10.1-py3.9.egg/acconeer/exptool/clients/links.py", line 234, in connect
> raise LinkError("failed to connect, timeout")
> acconeer.exptool.clients.links.LinkError: failed to connect, timeout
> sasha@Sashas-Mac acconeer-python-exploration-master %

That's quite odd - this error occurs before the configuration has any effect on the communication. After you get this, can you immediately run the `envelope.py` script without issues, without restarting the EVK?

Ondobaka, 2021-05-10:
Just tried that - envelope runs just fine straight after getting the breathing error.

erikmansson, 2021-05-16:
Sorry, but I am stumped on this one. There _shouldn't_ be any difference between the examples up to the point of the error. Do you still experience this issue?

closed 2021-05-26 by erikmansson